MIT-SPARK/Hydra #2: Issue with custom vio install

duda1202, 2022-06-30:
Hi,

I would like to know if the vio_overlay.rosinstall is only available for mit users or for the public too? As I have tried to run the script but it says permission denied. Thank you.

nathanhhughes, 2022-06-30:
Thanks for your interest in our work! As mentioned in the README, the vio_overlay.rosinstall is **not** publicly available at the moment (that particular overlay file is for ease of internal development for SparkLab). We're working on preparing a new release of Kimera-VIO (and Kimera-VIO-ROS) that should include the Hydra specific features, hopefully in the next few months. However, those branches are **not required** for running Kimera-VIO and Hydra (I'll update the README to be a little clearer on that point). Instead, they allow Kimera-VIO to publish the BoW vectors for every keyframe using [this](https://github.com/MIT-SPARK/pose_graph_tools/blob/feature/hydra/msg/BowQuery.msg) message, and a pose graph using [this](https://github.com/MIT-SPARK/pose_graph_tools/blob/feature/hydra/msg/PoseGraph.msg) message for use in the loop closure detection module.

duda1202, 2022-06-30:
So the DGS module works in the current Kimera-VIO/Semantics? I was trying to map my custom data and, at the moment, it leaves a trace of the dynamic entities on the map.

nathanhhughes, 2022-07-06:
I can confirm that the dynamic masking feature in Kimera-Semantics does appear to work for uHumans2 (I tested against the [uHumans2_office_s1_06h.bag](https://drive.google.com/uc?id=1zECekG47mlGafaJ84vCbwcx3Dz03NuvN)): 
[image]

I also put together an alternative rosinstall file that points to a tested branch of Kimera-VIO that can be compiled in the same workspace in Hydra and runs against uHumans2, which I'll merge in momentarily.  This still doesn't include features that Hydra uses (i.e. publishing the pose graph and loop closure information).

PR #5 (2022-07-06): Feature/public kimera
closes #2 by adding a temporary rosinstall for the public Kimera-VIO version.