MIT-SPARK/Hydra #6: [incremental_dsg_builder_node-12] has died! process has died! -- A bug on ubuntu 20.04 and ROS Noetic

Hao-Starrr, 2022-07-09:
Hi, 
I am an undergraduate at CUHK, working on my summer research. I am really impressed by Hydra's excellent outcome and novel ideas. 

But when I am trying to re-implement the code, I faced an unforeseen bug. I followed the instructions in readme file. After building it, I opened `roscore` and tried to run `roslaunch hydra_dsg_builder uhumans2_incremental_dsg.launch start_visualizer:=true`. It seemed that `[incremental_dsg_builder_node-12]` occurs the problem. Rviz flashed once and disappeared. 

The error message was like this:

```
[ INFO] [1657355430.505268029]: Initializing pose graph visualizer
[ WARN] [1657355430.507929902]: PoseGraphPublisher waiting for subscriber...
[ INFO] [1657355430.564287765]: rviz version 1.14.14
[ INFO] [1657355430.564318708]: compiled against Qt version 5.12.8
[ INFO] [1657355430.564326571]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1657355430.579810715]: Forcing OpenGl version 0.
[ INFO] [1657355430.736187684]: Stereo is NOT SUPPORTED
[ INFO] [1657355430.736225276]: OpenGL device: Mesa Intel(R) UHD Graphics 630 (CFL GT2)
[ INFO] [1657355430.736242671]: OpenGl version: 4.6 (GLSL 4.6) limited to GLSL 1.4 on Mesa system.
================================================================================
REQUIRED process [incremental_dsg_builder_node-12] has died!
process has died [pid 32014, exit code -11, cmd /home/haostarrr/catkin_ws/devel/lib/hydra_dsg_builder/hydra_dsg_builder_incremental_node --minloglevel=3 -v=0 ~voxblox_mesh:=/hydra_topology_node/active_mesh ~active_places:=/hydra_topology_node/active_layer ~/bow_vectors:=/kimera_vio_ros/bow_query frame_registration:=/kimera_vio_ros/kimera_vio_ros_node/register_lcd_frames __name:=incremental_dsg_builder_node __log:=/home/haostarrr/.ros/log/0c6383cc-ff61-11ec-8540-f5c17cd13c52/incremental_dsg_builder_node-12.log].
log file: /home/haostarrr/.ros/log/0c6383cc-ff61-11ec-8540-f5c17cd13c52/incremental_dsg_builder_node-12*.log
Initiating shutdown!
================================================================================
[incremental_dsg_builder_node/pgmo/posegraph_viewer-14] killing on exit
[pose_graph_publisher_node-13] killing on exit
[incremental_dsg_builder_node-12] killing on exit
[rviz-11] killing on exit
[hydra_visualizer_node-10] killing on exit
[bag_static_tf_4-5] killing on exit
[cloudify-8] killing on exit
[nodelet_manager-7] killing on exit
[bag_static_tf_5-6] killing on exit
[hydra_topology_node-9] killing on exit
[bag_static_tf_3-4] killing on exit
[bag_static_tf_2-3] killing on exit
[bag_static_tf_1-2] killing on exit
timing stats: elapsed: 0 [s] (0 +/- 0 [s] with 0 measurements)
mesh timing stats: elapsed: 0 [s] (0 +/- 0 [s] with 0 measurements)
terminate called after throwing an instance of 'boost::wrapexcept<boost::lock_error>'
  what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument
[bag_static_tf_0-1] killing on exit
QFileSystemWatcher::removePaths: list is empty
QFileSystemWatcher::removePaths: list is empty
shutting down processing monitor...
... shutting down processing monitor complete
done
```

I am using Ubuntu 20.04 and ROS Noetic. I am not sure whether it is my ROS's problem. I run ORB-SLAM3 last week (it may prove that my ROS can work?). 
My `.bashrc` setting is like:

```
source /opt/ros/noetic/setup.bash

# For ORB-SLAM3
# export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:/home/haostarrr/ORB_SLAM3_detailed_comments/Examples/ROS/ORB_SLAM3
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:/home/haostarrr/ORB_SLAM3/Examples_old/ROS/ORB_SLAM3

# For Hydra catkin_ws
source /home/haostarrr/catkin_ws/devel/setup.bash
```

Do you have any idea about this?

Thank you.

nathanhhughes, 2022-07-09:
Thank you for your interest in our work, and for taking the time to write a bug report!  I have a couple ideas as to what might be causing this, but the following pieces of information would help narrow things down:

**Running hydra with GDB to get a backtrace:** The most helpful piece of information for me would be to see a backtrace of where the segfault is occurring. You may have to install GDB (`sudo apt install gdb`) if you haven't already. You can then run:

```
roslaunch hydra_dsg_builder start_visualizer:=false debug:=true
``` 

Running with the visualizer doesn't really matter (i.e. you could run with `start_visualizer:=true`), it just makes the output logs cleaner to not run with rviz and the visualizer. You'll should see something like this: 
[image]

Enter `run` into the prompt (type `run` and then hit enter), and Hydra should attempt to start up (it might take a minute or so). It should crash immediately (if I'm understanding what's causing the issue correctly), or you may need to play the rosbag for a little bit to replicate the crash. Once it does crash, enter `bt` into the promt and you should get output similar to this (though this backtrace is from an incomplete feature I'm currently working on):

```
Thread 1 "hydra_dsg_build" received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
51      ../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007ffff5a547f1 in __GI_abort () at abort.c:79
#2  0x00007ffff6f234e7 in ?? () from /usr/lib/x86_64-linux-gnu/libglog.so.0
#3  0x00007ffff6f1a0cd in google::LogMessage::Fail() () from /usr/lib/x86_64-linux-gnu/libglog.so.0
#4  0x00007ffff6f1bf33 in google::LogMessage::SendToLog() () from /usr/lib/x86_64-linux-gnu/libglog.so.0
#5  0x00007ffff6f19c28 in google::LogMessage::Flush() () from /usr/lib/x86_64-linux-gnu/libglog.so.0
#6  0x00007ffff6f1c999 in google::LogMessageFatal::~LogMessageFatal() () from /usr/lib/x86_64-linux-gnu/libglog.so.0
#7  0x00007ffff521cb83 in kimera::SemanticLabel2Color::SemanticLabel2Color (this=0x555555811340, filename=...) at /home/ubuntu/catkin_ws/src/kimera_semantics/kimera_semantics/src/color.cpp:51
#8  0x00007ffff7b1dbbb in hydra::incremental::DsgFrontend::DsgFrontend (this=0x7fffffffcb40, nh=..., dsg=...) at /home/ubuntu/catkin_ws/src/hydra/hydra_dsg_builder/src/incremental_dsg_frontend.cpp:58
#9  0x0000555555560bfe in run (nh=..., frontend_dsg=std::shared_ptr<hydra::incremental::SharedDsgInfo> (use count 3, weak count 0) = {...}, 
    backend_dsg=std::shared_ptr<hydra::incremental::SharedDsgInfo> (use count 2, weak count 0) = {...}, output_path="/home/ubuntu/catkin_ws/src/hydra/hydra_dsg_builder/output/uhumans2/office")
    at /home/ubuntu/catkin_ws/src/hydra/hydra_dsg_builder/src/incremental_dsg_builder_node.cpp:56
#10 0x000055555556061d in main (argc=<optimized out>, argv=<optimized out>) at /home/ubuntu/catkin_ws/src/hydra/hydra_dsg_builder/src/incremental_dsg_builder_node.cpp:123
(gdb) 
```` 

Attaching this backtrace would be really helpful.

**Other Useful Information**
It'd also help to have your catkin workspace configuration:
```
roscd
catkin config
```
and the repository information from your workspace: 
```
roscd && cd ..
vcs export src > workspace_info.yaml
```

Hao-Starrr, 2022-07-09:
Hi

After I entered `run`, it shows:

```
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/haostarrr/catkin_ws/devel/lib/hydra_dsg_builder/hydra_dsg_builder_incremental_node...
(No debugging symbols found in /home/haostarrr/catkin_ws/devel/lib/hydra_dsg_builder/hydra_dsg_builder_incremental_node)
(gdb) run
Starting program: /home/haostarrr/catkin_ws/devel/lib/hydra_dsg_builder/hydra_dsg_builder_incremental_node --minloglevel=3 -v=0 \~voxblox_mesh:=/hydra_topology_node/active_mesh \~active_places:=/hydra_topology_node/active_layer \~/bow_vectors:=/kimera_vio_ros/bow_query frame_registration:=/kimera_vio_ros/kimera_vio_ros_node/register_lcd_frames __name:=incremental_dsg_builder_node __log:=/home/haostarrr/.ros/log/660f6410-ff91-11ec-8540-f5c17cd13c52/incremental_dsg_builder_node-10.log
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Program received signal SIGSEGV, Segmentation fault.
0x00007fffee0da5d1 in ?? ()
   from /lib/x86_64-linux-gnu/libpcl_sample_consensus.so.1.10
(gdb) 

```
No crash yet. 
Then I started to play the bag. I got this but stuck at time 3.640108. Then nothing happened. 

```
haostarrr@haostarrr-ROG-Strix-G531GU-S5DU:~$ rosbag play /home/haostarrr/Datasets/uHumans2/uHumans2_apartment_s1_00h.orig.bag --clock
[ INFO] [1657377130.648450250]: Opening /home/haostarrr/Datasets/uHumans2/uHumans2_apartment_s1_00h.orig.bag

Waiting 0.2 seconds after advertising topics... done.

Hit space to toggle paused, or 's' to step.
 [RUNNING]  Bag Time:      3.500000   Duration: 0.000000 / 138.397900            
[RUNNING]  Bag Time:      3.500197   Duration: 0.000197 / 138.397900            
[RUNNING]  Bag Time:      3.505161   Duration: 0.005161 / 138.397900            
[RUNNING]  Bag Time:      3.510262   Duration: 0.010262 / 138.397900            
[RUNNING]  Bag Time:      3.515154   Duration: 0.015154 / 138.397900            
[RUNNING]  Bag Time:      3.520328   Duration: 0.020328 / 138.397900            
[RUNNING]  Bag Time:      3.525172   Duration: 0.025172 / 138.397900            
[RUNNING]  Bag Time:      3.530393   Duration: 0.030393 / 138.397900            
[RUNNING]  Bag Time:      3.535181   Duration: 0.035181 / 138.397900            
[RUNNING]  Bag Time:      3.540460   Duration: 0.040460 / 138.397900            
[RUNNING]  Bag Time:      3.545168   Duration: 0.045168 / 138.397900            
[RUNNING]  Bag Time:      3.627211   Duration: 0.127211 / 138.397900            
[RUNNING]  Bag Time:      3.630095   Duration: 0.130095 / 138.397900            
[RUNNING]  Bag Time:      3.637268   Duration: 0.137268 / 138.397900            
[RUNNING]  Bag Time:      3.640108   Duration: 0.140108 / 138.397900               
```
I got the bag from [here](http://web.mit.edu/sparklab/datasets/uHumans2/) and decompress it. The downloaded file is `uHumans2_apartment_s1_00h.bag` (9.3G). After decompressing it, the name become `uHumans2_apartment_s1_00h.orig.bag` (2.7G). Looks weird here. 

I tried to input `run` again and:
```
(gdb) run
The program being debugged has been started already.
Start it from the beginning? (y or n) n
Program not restarted.
```
Try input `bt`:

```
(gdb) bt
#0  0x00007fffee0da5d1 in ?? ()
   from /lib/x86_64-linux-gnu/libpcl_sample_consensus.so.1.10
#1  0x00007ffff7fe0b9a in call_init (l=<optimized out>, argc=argc@entry=9, 
    argv=argv@entry=0x7fffffffd408, env=env@entry=0x7fffffffd458)
    at dl-init.c:72
#2  0x00007ffff7fe0ca1 in call_init (env=0x7fffffffd458, argv=0x7fffffffd408, 
    argc=9, l=<optimized out>) at dl-init.c:30
#3  _dl_init (main_map=0x7ffff7ffe190, argc=9, argv=0x7fffffffd408, 
    env=0x7fffffffd458) at dl-init.c:119
#4  0x00007ffff7fd013a in _dl_start_user () from /lib64/ld-linux-x86-64.so.2
#5  0x0000000000000009 in ?? ()
[... 93 lines cut ...]
  minkindr:
    type: git
    url: git@github.com:ethz-asl/minkindr.git
    version: master
  minkindr_ros:
    type: git
    url: git@github.com:ethz-asl/minkindr_ros.git
    version: master
  pose_graph_tools:
    type: git
    url: git@github.com:MIT-SPARK/pose_graph_tools.git
    version: feature/hydra
  spark_dsg:
    type: git
    url: git@github.com:MIT-SPARK/Spark-DSG.git
    version: main
  teaser_plusplus:
    type: git
    url: git@github.com:MIT-SPARK/TEASER-plusplus.git
    version: master
  voxblox:
    type: git
    url: git@github.com:MIT-SPARK/voxblox.git
    version: feature/hydra
```

nathanhhughes, 2022-07-09:
> No crash yet.

Hydra actually did crash once you started it (i.e. `Program received signal SIGSEGV, Segmentation fault.`), and playing the rosbag didn't change things. It looks like the crash happens when trying to load in `libpcl_sample_consensus.so.1.10.0` (i.e. one of the PCL modules that Hydra links against. However, I'm not entirely sure why.  There are a couple options I could think of:
- You installed PCL from source and enabled `-march=native` when building PCL
- Somehow something corrupted the PCL install for the sample consensus module
- You have a previous source install of PCL for something else that gets picked up when building Hydra and Hydra is building against a different version of the PCL headers than it is linking against

Unless the first option is true (i.e. you installed PCL from source), the next debug step is to rebuild `hydra_dsg_builder` and grab the output. To do this:

```
roscd && cd ..
catkin clean hydra_dsg_builder
catkin build hydra_dsg_builder --no-deps -v
```
and then attach the output. I'll also double check that a clean install of Hydra works on my side / see If I can limit how much of PCL gets linked into Hydra.

> I got the bag from [here](http://web.mit.edu/sparklab/datasets/uHumans2/) and decompress it. The downloaded file is uHumans2_apartment_s1_00h.bag (9.3G). After decompressing it, the name become uHumans2_apartment_s1_00h.orig.bag (2.7G). Looks weird here.

`rosbag decompress` will save the original downloaded bag under `uHumans2_apartment_s1_00h.orig.bag` and the decompressed bag under the name of the downloaded bag (`uHumans2_apartment_s1_00h.bag`). The size difference between the original bag (2.7G) and the decompressed bag (9.3G) looks right, and the `rosbag play` behavior on the compressed bag (i.e. `uHumans2_apartment_s1_00h.orig.bag`) also seems right (playback will be significantly slower and stop occasionally on compressed rosbags). You should be able to just do `rosbag play uHumans2_apartment_s1_00h.bag` from now on (and you can get rid of the `uHumans2_apartment_s1_00h.orig.bag` if you'd like).

:warning: The apartment rosbag uses a different set of semantic colors than the recommended office rosbag. You'll have to specify the location of [this](https://github.com/MIT-SPARK/Kimera-Semantics/blob/master/kimera_semantics_ros/cfg/tesse_multiscene_archviz1_segmentation_mapping.csv) as an argument to Hydra as well as a different typology file. For your setup, it looks like it'd be:

```
roslaunch hydra_dsg_builder uhumans2_incremental_dsg.launch start_visualizer:=true semantic_color_path:=/home/haostarr/catkin_ws/src/kimera_semantics/kimera_semantics_ros/cfg.tesse_multiscene_archviz1_segmentation_mapping.csv typology_config:=uhumans2_apartment_typology.yaml
```

Hao-Starrr, 2022-07-10:
Yes! I did install PCL before! So when I rebuild `hydra_dsg_builder`, I got warnings about PCL. 

For rosbag, exactly the same as what you said. The compressed bag is very slow. 

The warnings are: 
```
haostarrr@haostarrr-ROG-Strix-G531GU-S5DU:~/catkin_ws/devel$ roscd && cd ..

haostarrr@haostarrr-ROG-Strix-G531GU-S5DU:~/catkin_ws$ catkin clean hydra_dsg_builder

Starting  >>> hydra_dsg_builder             
Finished  <<< hydra_dsg_builder              [ 0.1 seconds ]
[clean] Summary: All 1 packages succeeded!
[clean]   Ignored:   26 packages were skipped or are skiplisted.
[clean]   Warnings:  None.
[clean]   Abandoned: None.
[clean]   Failed:    None.
[clean] Runtime: 1.0 seconds total.
[clean] Note: Parts of the workspace have been cleaned which will necessitate re-configuring CMake on the next build.

haostarrr@haostarrr-ROG-Strix-G531GU-S5DU:~/catkin_ws$ catkin build hydra_dsg_builder --no-deps -v

--------------------------------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/noetic
Workspace:                   /home/haostarrr/catkin_ws
--------------------------------------------------------------------------------
Build Space:        [exists] /home/haostarrr/catkin_ws/build
Devel Space:        [exists] /home/haostarrr/catkin_ws/devel
Install Space:      [unused] /home/haostarrr/catkin_ws/install
Log Space:          [exists] /home/haostarrr/catkin_ws/logs
Source Space:       [exists] /home/haostarrr/catkin_ws/src
DESTDIR:            [unused] None
--------------------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
--------------------------------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release -DGTSAM_TANGENT_PREINTEGRATION=OFF -DGTSAM_BUILD_WITH_MARCH_NATIVE=OFF -DOPENGV_BUILD_WITH_MARCH_NATIVE=OFF
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
--------------------------------------------------------------------------------
Buildlisted Packages:        None
Skiplisted Packages:         hdf5_map_io mesh_msgs_hdf5 label_manager mesh_tools rviz_map_plugin minkindr_python
--------------------------------------------------------------------------------
Workspace configuration appears valid.

NOTE: Forcing CMake to run for each package.
--------------------------------------------------------------------------------
[build] Found 32 packages in 0.0 seconds.                                      
[build] Package table is up to date.                                           
Starting  >>> hydra_dsg_builder                                                
Starting   >> hydra_dsg_builder:loadenv                                        
Output     << hydra_dsg_builder:loadenv /home/haostarrr/catkin_ws/logs/hydra_dsg_builder/build.loadenv.001.log
Loading environment from: /home/haostarrr/catkin_ws/devel/env.sh
              
Finished   << hydra_dsg_builder:loadenv                                        
Starting   >> hydra_dsg_builder:mkdir                                          
Starting   >> hydra_dsg_builder:mkdir                                          
Starting   >> hydra_dsg_builder:cache-manifest                                 
Starting   >> hydra_dsg_builder:cmake                                          
Subprocess  > hydra_dsg_builder:cmake `cd /home/haostarrr/catkin_ws/build/hydra_dsg_builder; catkin build --get-env hydra_dsg_builder | catkin env -si  /usr/bin/cmake /home/haostarrr/catkin_ws/src/hydra/hydra_dsg_builder --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/haostarrr/catkin_ws/devel/.private/hydra_dsg_builder -DCMAKE_INSTALL_PREFIX=/home/haostarrr/catkin_ws/install -DCMAKE_BUILD_TYPE=Release -DGTSAM_TANGENT_PREINTEGRATION=OFF -DGTSAM_BUILD_WITH_MARCH_NATIVE=OFF -DOPENGV_BUILD_WITH_MARCH_NATIVE=OFF; cd -`
_______________________________________________________________________________
Warnings   << hydra_dsg_builder:cmake /home/haostarrr/catkin_ws/logs/hydra_dsg_builder/build.cmake.001.log
Not searching for unused variables given on the command line.
Re-run cmake no build system arguments
-- The C compiler identification is GNU 9.4.0
-- The CXX compiler identification is GNU 9.4.0
-- Check for working C compiler: /usr/bin/cc
[... 480 lines cut ...]

CMake Warning at /usr/share/cmake-3.16/Modules/FindBoost.cmake:1161 (message):
  New Boost version may have incorrect or missing dependencies and imported
  targets
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/FindBoost.cmake:1283 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/share/cmake-3.16/Modules/FindBoost.cmake:1921 (_Boost_MISSING_DEPENDENCIES)
  /usr/share/cmake-3.16/Modules/CMakeFindDependencyMacro.cmake:47 (find_package)
  /home/haostarrr/catkin_ws/devel/lib/cmake/GTSAM_UNSTABLE/GTSAM_UNSTABLEConfig.cmake:21 (find_dependency)
  /home/haostarrr/catkin_ws/devel/lib/cmake/KimeraRPGO/KimeraRPGOConfig.cmake:14 (find_package)
  CMakeLists.txt:27 (find_package)

CMake Warning at /usr/share/cmake-3.16/Modules/FindBoost.cmake:1161 (message):
  New Boost version may have incorrect or missing dependencies and imported
  targets
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/FindBoost.cmake:1283 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/share/cmake-3.16/Modules/FindBoost.cmake:1921 (_Boost_MISSING_DEPENDENCIES)
  /usr/share/cmake-3.16/Modules/CMakeFindDependencyMacro.cmake:47 (find_package)
  /home/haostarrr/catkin_ws/devel/lib/cmake/GTSAM_UNSTABLE/GTSAM_UNSTABLEConfig.cmake:21 (find_dependency)
  /home/haostarrr/catkin_ws/devel/lib/cmake/KimeraRPGO/KimeraRPGOConfig.cmake:14 (find_package)
  CMakeLists.txt:27 (find_package)
```

Then I checked `hydra.rosinstall` file. I didn't find PCL in it. Is that mean: PCL is a dependency of one repo in `hydra.rosinstall`?

nathanhhughes, 2022-07-11:
> Then I checked hydra.rosinstall file. I didn't find PCL in it. Is that mean: PCL is a dependency of one repo in hydra.rosinstall?

PCL ([the pointcloud library](https://pointclouds.org/)) is a dependency of a couple packages (e.g. `voxblox_ros`) and also a dependency of Hydra's (though I forgot to specify it directly in the `package.xml`). It's a system library though that gets pulled in on most ROS installs (the package itself is `libpcl-dev`).  

> The warnings are:

Looking at the output from cmake, it seems like the problem might be with a custom installation of boost on your machine instead; you have a lot of cmake warnings about missing dependencies in a newer version of boost than what 20.04 ships with. I see `/usr/local/lib/cmake/Boost-1.75.0/BoostConfig.cmake` is the version cmake is finding on your machine from your logs, whereas `/usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake` is the system version that I'm finding on my machine. I'd suggest trying one of the following:
- Removing the custom installation of boost and doing a clean build of the workspace
- Trying docker to avoid problems with your system (#3 discusses success using the `osrf/ros:noetic-desktop` docker image)
- Setting `CMAKE_SYSTEM_PREFIX_PATH` to find `/usr/` before `/usr/local/` (`catkin config -a --cmake-args [CUSTOM_PREFIX_PATH]` should make it straightforward to do that) and doing a clean build of the workspace

nathanhhughes, 2022-07-12:
Great, glad that you were able to get it to work!