MIT-SPARK/Hydra #9: Couldn't open the start_visualizer option

zhanghua7099, 2022-07-20:
Hi!

I found a bug that may be occurred when using the option: "start visualizer:=true". If I launched the Hydra by:
```
roslaunch hydra_dsg_builder uhumans2_incremental_dsg.launch start_visualizer:=true
```

The hydra will be killed and output the following message:
```
[ INFO] [1658319483.616856493]: Config: 
- mesh_plugin_type: PGMO
- load_graph: false
- scene_graph_filepath: 
- visualizer_ns: /hydra_dsg_visualizer
- mesh_plugin_ns: dsg_mesh
- output_path: 

[ WARN] [1658319483.626158232]: PoseGraphPublisher waiting for subscriber...
[ INFO] [1658319483.627042257]: Initializing pose graph visualizer
[ INFO] [1658319483.700555267]: rviz version 1.14.14
[ INFO] [1658319483.700585160]: compiled against Qt version 5.12.8
[ INFO] [1658319483.700590221]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1658319483.712954538]: Forcing OpenGl version 0.
[ INFO] [1658319483.714701817]: Logging backend graph to /home/zhy/ROSProject/hydar_ws/src/hydra/hydra_dsg_builder/output/uhumans2/office/backend
[ WARN] [1658319483.720185107]: MeshFrontend: Skipping callback registration
[ INFO] [1658319483.720202249]: Initialized MeshFrontend.
[ INFO] [1658319483.720210421]: Logging frontend graph to /home/zhy/ROSProject/hydar_ws/src/hydra/hydra_dsg_builder/output/uhumans2/office/frontend
[ WARN] [1658319484.029669728, 17.987247466]: PoseGraphPublisher waiting for transform: left_cam -> world
[ INFO] [1658319484.118245471, 18.077831406]: Stereo is NOT SUPPORTED
[ INFO] [1658319484.118289907, 18.077831406]: OpenGL device: NVIDIA GeForce RTX 3090/PCIe/SSE2
[ INFO] [1658319484.118303260, 18.077831406]: OpenGl version: 4.6 (GLSL 4.6).
[ WARN] [1658319484.227130625, 18.188874276]: using timer!
[ INFO] [1658319484.536664342, 18.500714565]: New prefix s detected when adding new mesh edges and nodes. 
[ WARN] [1658319484.536824797, 18.500714565]: KimeraPgmo: Partial mesh not connected to pose graph. 
[ INFO] [1658319484.537410575, 18.500714565]: DeformationGraph: Re-calculating all mesh vertices in deformMesh. 
[ INFO] [1658319484.538214830, 18.500714565]: Publish optimized mesh. 
[ INFO] [1658319484.547218824, 18.510770073]: Received dsg update message of 7.58 KiB
[ INFO] [1658319484.723557844, 18.681757309]: Creating TriangleMeshVisual 0_TriangleMesh_0_8400
[ INFO] [1658319484.723641811, 18.681757309]: Resetting TriangleMeshVisual 0_TriangleMesh_0_8400
[ INFO] [1658319484.723668978, 18.681757309]: Received 2066 vertex colors.
[ INFO] [1658319485.138061584, 19.094287005]: Publish optimized mesh. 
[ INFO] [1658319485.151557365, 19.114415800]: Received dsg update message of 9.86 KiB
*** Aborted at 1658319485 (unix time) try "date -d @1658319485" if you are using GNU date ***
PC: @                0x0 (unknown)
*** SIGSEGV (@0x100000001) received by PID 732470 (TID 0x7ff7647eb000) from PID 1; stack trace: ***
    @     0x7ff769846090 (unknown)
    @     0x7ff7695be7a3 cv::_OutputArray::create()
    @     0x7ff769df4a5e cv::cvtColorHLS2BGR()
    @     0x7ff769dea598 cv::cvtColor()
    @     0x7ff76bbafdda hydra::dsg_utils::getRgbFromHls()
    @     0x7ff76bb9c210 hydra::getNodeColor()
    @     0x7ff76bb9d124 hydra::DynamicSceneGraphVisualizer::drawDynamicLayer()
    @     0x7ff76bba0e40 hydra::DynamicSceneGraphVisualizer::drawDynamicLayers()
    @     0x7ff76bba1aad hydra::DynamicSceneGraphVisualizer::redrawImpl()
    @     0x7ff76bb9c671 hydra::DynamicSceneGraphVisualizer::redraw()
    @     0x55a805235414 hydra::VisualizerNode::spin()
    @     0x55a805232f3e main
    @     0x7ff769827083 __libc_start_main
    @     0x55a80523302e _start
================================================================================REQUIRED process [hydra_visualizer_node-10] has died!
process has died [pid 732470, exit code -11, cmd /home/zhy/ROSProject/hydar_ws/devel/lib/hydra_utils/hydra_visualizer_node -alsologtostderr -colorlogtostderr -v=0 ~dsg:=/incremental_dsg_builder_node/dsg ~dsg_mesh_updates:=/incremental_dsg_builder_node/pgmo/optimized_mesh __name:=hydra_visualizer_node __log:=/home/zhy/.ros/log/fad91482-0825-11ed-85c5-efe60e99c719/hydra_visualizer_node-10.log].
log file: /home/zhy/.ros/log/fad91482-0825-11ed-85c5-efe60e99c719/hydra_visualizer_node-10*.log
Initiating shutdown!
================================================================================
[incremental_dsg_builder_node/pgmo/posegraph_viewer-14] killing on exit
[pose_graph_publisher_node-13] killing on exit
[incremental_dsg_builder_node-12] killing on exit
[rviz-11] killing on exit
[hydra_visualizer_node-10] killing on exit
[hydra_topology_node-9] killing on exit
[cloudify-8] killing on exit
[nodelet_manager-7] killing on exit
[bag_static_tf_3-4] killing on exit
[bag_static_tf_4-5] killing on exit
[bag_static_tf_5-6] killing on exit
[bag_static_tf_2-3] killing on exit
[bag_static_tf_1-2] killing on exit
[bag_static_tf_0-1] killing on exit
QFileSystemWatcher::removePaths: list is empty
QFileSystemWatcher::removePaths: list is empty
shutting down processing monitor...
... shutting down processing monitor complete
done
```

If I didn't use the option "start_visualizer:=true", the hydra runs well:
```
roslaunch hydra_dsg_builder uhumans2_incremental_dsg.launch
```

The hydra's output can be also seen by selecting the topics in Rviz. It may be that some bugs in visualization. How to solve it?

THX!

nathanhhughes, 2022-07-20:
Hi, thank you for your interest in our work! From the backtrace, it looks like the problem might be with your opencv installation rather than anything the visualizer is doing (i.e. it looks like the code is segfaulting when trying to allocate a 3 element `cv::Mat` to store a color). Couple initial questions:
- What ubuntu version are you on?
- Have you installed a custom version of opencv / is hydra being built against an opencv version that isn't the default system version?

> hydra's output can be also seen by selecting the topics in Rviz.

You're saying that you can add the mesh from voxblox or other visualization topics from the `hydra_dsg_builder` node to rviz and see them visualized fine, correct? But you still can't actually visualize the scene graph itself (i.e. because the visualizer node segfaults)?

zhanghua7099, 2022-07-21:
Thank you for your help! 

Yes, I installed OpenCV 4.5.4 (not the default version in Ubuntu 20.04). After I uninstalled it and used the default version, the hydra works well!

closed 2022-07-21 by zhanghua7099

qingyun322, 2022-07-27:
Hi @zhanghua7099 what's the version of opencv you are using? I encountered the same issue but some how I can't find the default opencv installed on my ubuntu.