MIT-SPARK/Hydra #10: Running Hydra with EuRoC dataset

MikeDmytrenko, 2022-07-25:
I managed to successfully build hydra with Kimear-VIO and it runs well on uHumans data.  How can I run it with EuRoc dataset? What has to change in the config file? Thanks!

nathanhhughes, 2022-07-25:
Hi, thanks for your interest in our work! Running on EuRoC is not currently supported for Hydra and we unfortunately do not plan on supporting it in the future either. Unlike uHumans2, EuRoC doesn't have available ground truth 2D semantic segmentation, so you would have to provide that yourself. You can see #1 for a general overview of how to set up Hydra with a custom real-world dataset (which applies for EuRoC in this case), including what you would need to change for a custom 2D semantic segmentation network.

That said, I'd recommend against investing time getting Hydra to work with EuRoC as the dataset is not well suited for Hydra: all three scenes in EuRoC (both vicon rooms and the machine hall) are single-room scenes with few if any objects, so the resulting scene graphs would be pretty boring. If you do end up trying to get things to work with EuRoC and have additional questions about configuration of Hydra, I'd kindly request that you ask them through issue #1 if relevant (so that other people benefit from having information about how to set up a custom dataset in the same place).

MikeDmytrenko, 2022-07-26:
Okay, thanks! I continued on the issue you mentioned

nathanhhughes, 2022-07-27:
You're welcome! I'm going to close this for now. If you end up having follow-up questions specific to EuRoC, feel free to reopen the issue.

closed 2022-07-27 by nathanhhughes