MIT-SPARK/Hydra #18: [QUESTION] running uHumans2 dataset against other scenes produces no objects

hwan0806, 2022-10-01:
First, thank you for wonderful work. 

I have a similar issue adapting uHuman2_apartment or subway dataset. 
When adapting apartment data, i exactly followed the instruction of uHumans2_office_s1_00h version and  the mapping result was similar with office case except meshed map. 

[image]

I thinked some parameters in `uhumans2_incremental_dsg.launch`  should be changed for other datasets and convert below arguments to appropriate one in  `kimera_semantics_ros/cfg` folder.

https://github.com/MIT-SPARK/Hydra/blob/7ccbadeb14cf73786e4ac88de7ef9a57cdb7459e/hydra_dsg_builder/launch/uhumans2_incremental_dsg.launch#L5-L9
https://github.com/MIT-SPARK/Hydra/blob/7ccbadeb14cf73786e4ac88de7ef9a57cdb7459e/hydra_topology/launch/hydra_topology_uhumans2.launch#L12-L14

After that, hydra package is not launched for below error messages. 

```
[ INFO] [1664551953.114403717]: Initializing pose graph visualizer
[ WARN] [1664551953.116088151]: PoseGraphPublisher waiting for subscriber...
F1001 00:32:33.176373 328674 color.cpp:51] Check failed: loop->size() == 6 (8 vs. 6) Row 1 is invalid.
*** Check failure stack trace: ***
    @     0x7f793f6ad1c3  google::LogMessage::Fail()
[ INFO] [1664551953.179036916]: Logging backend graph to /home/gil/sparoLab/open_source/Graph_slam/src/hydra/hydra_dsg_builder/output/uhumans2/apartment/backend
    @     0x7f793f6b225b  google::LogMessage::SendToLog()
    @     0x7f793f6acebf  google::LogMessage::Flush()
    @     0x7f793f6ad6ef  google::LogMessageFatal::~LogMessageFatal()
    @     0x7f793e3073ab  kimera::SemanticLabel2Color::SemanticLabel2Color()
    @     0x7f793faee0e6  kimera::getSemanticTsdfIntegratorConfigFromRosParam()
    @     0x7f793faf8cb5  kimera::SemanticTsdfServer::SemanticTsdfServer()
[ INFO] [1664551953.183147875]: Logging output to: /home/gil/sparoLab/open_source/Graph_slam/src/hydra/hydra_dsg_builder/output/uhumans2/apartment/pgmo
    @     0x7f793faf945e  kimera::SemanticTsdfServer::SemanticTsdfServer()
    @     0x56209002d449  hydra::topology::TopologyServer<>::setupLayers()
    @     0x56209002dd64  hydra::topology::TopologyServer<>::TopologyServer()
    @     0x562090017a3b  main
    @     0x7f793f1b7083  __libc_start_main
    @     0x562090017c9e  _start
[ WARN] [1664551953.184279743]: MeshFrontend: Skipping callback registration
[ INFO] [1664551953.185095673]: Initialized MeshFrontend.
[ INFO] [1664551953.185166448]: Logging frontend graph to /home/gil/sparoLab/open_source/Graph_slam/src/hydra/hydra_dsg_builder/output/uhumans2/apartment/frontend
[ INFO] [1664551953.187218466]: rviz version 1.14.19
[ INFO] [1664551953.187237556]: compiled against Qt version 5.12.8
[ INFO] [1664551953.187241696]: compiled against OGRE version 1.9.0 (Ghadamon)
F1001 00:32:33.199155 328688 color.cpp:51] Check failed: loop->size() == 6 (8 vs. 6) Row 1 is invalid.
*** Check failure stack trace: ***
[ INFO] [1664551953.200407565]: Forcing OpenGl version 0.
    @     0x7f41c032e1c3  google::LogMessage::Fail()
    @     0x7f41c033325b  google::LogMessage::SendToLog()
    @     0x7f41c032debf  google::LogMessage::Flush()
    @     0x7f41c032e6ef  google::LogMessageFatal::~LogMessageFatal()
    @     0x7f41bf6693ab  kimera::SemanticLabel2Color::SemanticLabel2Color()
    @     0x7f41bf6a80e6  kimera::getSemanticTsdfIntegratorConfigFromRosParam()
    @     0x7f41c0980035  hydra::incremental::MeshSegmenter::MeshSegmenter()
    @     0x7f41c0966aa8  hydra::incremental::DsgFrontend::startMeshFrontend()
    @     0x7f41c0967c9a  hydra::incremental::DsgFrontend::start()
    @     0x55ec45473f57  main
    @     0x7f41bf7ed083  __libc_start_main
    @     0x55ec4547496e  _start
================================================================================REQUIRED process [hydra_topology_node-10] has died!
process has died [pid 328674, exit code -6, cmd /home/gil/sparoLab/open_source/Graph_slam/devel/lib/hydra_topology/hydra_topology_node --minloglevel=3 -v=0 pointcloud:=/semantic_pointcloud __name:=hydra_topology_node __log:=/home/gil/.ros/log/19f051fe-40d5-11ed-a742-e51d644fcd44/hydra_topology_node-10.log].
log file: /home/gil/.ros/log/19f051fe-40d5-11ed-a742-e51d644fcd44/hydra_topology_node-10*.log
Initiating shutdown!
================================================================================
```

Could i get some advise for these issues?

Thanks

hwan0806, 2022-10-01:
https://github.com/MIT-SPARK/Hydra/issues/17#issuecomment-1263733237

nathanhhughes, 2022-10-04:
Thanks for your interest!

Yes, it should be:

```
roslaunch hydra_dsg_builder uhumans2_incremental_dsg.launch start_visualizer:=true semantic_map_file:=tesse_multiscene_archviz1_segmentation_mapping.csv typology_file:=uhmans2_apartment_typology.yaml
```

**An explanation:** Unfortunately, the mapping between colors and semantic categories (and available semantic categories) changes between different uHumans2 scenes, and Kimera-Semantics (which performs the 3D metric-semantic reconstruction) needs to be pointed to a new csv file.  This is the reason that you were getting was a mesh that was entirely white and not extracting any objects: the colors between the office and the apartment scene don't overlap at all and the launch files default to the office scene colors. The name for the apartment scene in the Unity simulator is "archviz", which is why the mapping file for the apartment is a little hard to find (your crash came from pointing Kimera-Semantics to a different kind of file).  

The second argument, the typology file, is to change the semantic categories that Hydra extracts objects for. It's not strictly necessary (you'll still get objects if you use the default object one), but does match the scene better.

hwan0806, 2022-10-05:
Thank you for kind and specific explanation!
All the issues are solved!

closed 2022-10-05 by hwan0806