MIT-SPARK/Hydra #20: [QUESTION]Dependencies about pcl and opencv's flann lib？

fengdelin, 2022-11-10:
**I tried adding a flann header or specifying a namespace, but nothing could fix the compilation error. Please help me, thank you.**

> /usr/include/pcl-1.8/pcl/kdtree/impl/kdtree_flann.hpp:123:55: **error**: expected primary-expression before ‘float’
   flann_index_.reset (new FLANNIndex (::flann::Matrix<float> (cloud_.get (),
                                                       ^~~~~
/usr/include/pcl-1.8/pcl/kdtree/impl/kdtree_flann.hpp:126:48: **error**: ‘KDTreeSingleIndexParams’ is not a member of ‘flann’
                                       ::flann::KDTreeSingleIndexParams (15))); // max 15 points/leaf
                                                ^~~~~~~~~~~~~~~~~~~~~~~
/usr/include/pcl-1.8/pcl/kdtree/impl/kdtree_flann.hpp:126:48: note: suggested alternative:
In file included from /usr/local/include/opencv2/flann/all_indices.h:37:0,
                 from /usr/local/include/opencv2/flann/flann_base.hpp:43,
                 from /usr/local/include/opencv2/flann.hpp:48,
                 from /usr/local/include/opencv2/flann/flann.hpp:48,
                 from /usr/include/pcl-1.8/pcl/kdtree/flann.h:50,
                 from /usr/include/pcl-1.8/pcl/kdtree/kdtree_flann.h:45,
                 from /usr/include/pcl-1.8/pcl/search/kdtree.h:44,
                 from /usr/include/pcl-1.8/pcl/search/pcl_search.h:44,
                 from /usr/include/pcl-1.8/pcl/features/impl/feature.hpp:44,
                 from /usr/include/pcl-1.8/pcl/features/feature.h:498,
                 from /usr/include/pcl-1.8/pcl/features/moment_of_inertia_estimation.h:45,
                 from /home/username/catkin_ws/devel/include/spark_dsg/bounding_box.h:37,
                 from /home/username/catkin_ws/devel/include/spark_dsg/node_attributes.h:36,
                 from /home/username/catkin_ws/devel/include/spark_dsg/scene_graph_node.h:36,
                 from /home/username/catkin_ws/devel/include/spark_dsg/base_layer.h:38,
                 from /home/username/catkin_ws/devel/include/spark_dsg/dynamic_scene_graph_layer.h:36,
                 from /home/username/catkin_ws/devel/include/spark_dsg/dynamic_scene_graph.h:36,
                 from /home/username/catkin_ws/src/hydra/hydra_utils/include/hydra_utils/dsg_types.h:2,
                 from /home/username/catkin_ws/src/hydra/hydra_utils/include/hydra_utils/dsg_visualizer_plugin.h:37,
                 from /home/username/catkin_ws/src/hydra/hydra_utils/include/hydra_utils/dsg_mesh_plugins.h:36,
                 from /home/username/catkin_ws/src/hydra/hydra_utils/src/dsg_mesh_plugins.cpp:35:
/usr/local/include/opencv2/flann/kdtree_single_index.h:51:8: note:   ‘cvflann::KDTreeSingleIndexParams’
 struct KDTreeSingleIndexParams : public IndexParams
        ^~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/pcl-1.8/pcl/kdtree/kdtree_flann.h:238:0,
                 from /usr/include/pcl-1.8/pcl/search/kdtree.h:44,
                 from /usr/include/pcl-1.8/pcl/search/pcl_search.h:44,
                 from /usr/include/pcl-1.8/pcl/features/impl/feature.hpp:44,
                 from /usr/include/pcl-1.8/pcl/features/feature.h:498,
                 from /usr/include/pcl-1.8/pcl/features/moment_of_inertia_estimation.h:45,
                 from /home/username/catkin_ws/devel/include/spark_dsg/bounding_box.h:37,
                 from /home/username/catkin_ws/devel/include/spark_dsg/node_attributes.h:36,
                 from /home/username/catkin_ws/devel/include/spark_dsg/scene_graph_node.h:36,
                 from /home/username/catkin_ws/devel/include/spark_dsg/base_layer.h:38,
                 from /home/username/catkin_ws/devel/include/spark_dsg/dynamic_scene_graph_layer.h:36,
                 from /home/username/catkin_ws/devel/include/spark_dsg/dynamic_scene_graph.h:36,
                 from /home/username/catkin_ws/src/hydra/hydra_utils/include/hydra_utils/dsg_types.h:2,
                 from /home/username/catkin_ws/src/hydra/hydra_utils/include/hydra_utils/dsg_visualizer_plugin.h:37,
                 from /home/username/catkin_ws/src/hydra/hydra_utils/include/hydra_utils/dsg_mesh_plugins.h:36,
                 from /home/username/catkin_ws/src/hydra/hydra_utils/src/dsg_mesh_plugins.cpp:35:
/usr/include/pcl-1.8/pcl/kdtree/impl/kdtree_flann.hpp: In member function ‘int pcl::KdTreeFLANN<PointT, Dist>::nearestKSearch(const PointT&, int, std::vector<int>&, std::vector<float>&) const’:
/usr/include/pcl-1.8/pcl/kdtree/impl/kdtree_flann.hpp:147:12: **error**: ‘Matrix’ is not a member of ‘flann’
   ::flann::Matrix<int> k_indices_mat (&k_indices[0], 1, k);
            ^~~~~~
/usr/include/pcl-1.8/pcl/kdtree/impl/kdtree_flann.hpp:147:12: note: suggested alternatives:
In file included from /usr/include/eigen3/Eigen/Core:346:0,
                 from /home/username/catkin_ws/devel/include/spark_dsg/scene_graph_types.h:36,
                 from /home/username/catkin_ws/devel/include/spark_dsg/node_symbol.h:36,
                 from /home/username/catkin_ws/devel/include/spark_dsg/edge_container.h:37,
                 from /home/username/catkin_ws/devel/include/spark_dsg/base_layer.h:37,
                 from /home/username/catkin_ws/devel/include/spark_dsg/dynamic_scene_graph_layer.h:36,
                 from /home/username/catkin_ws/src/hydra/hydra_utils/include/hydra_utils/dsg_mesh_plugins.h:36,
                 from /home/username/catkin_ws/src/hydra/hydra_utils/src/dsg_mesh_plugins.cpp:35:
/usr/include/eigen3/Eigen/src/Core/util/ForwardDeclarations.h:74:9: note:   ‘Eigen::Matrix’ 

 > class Matrix;
         ^~~~~~
In file included from /usr/local/include/opencv2/flann/flann_base.hpp:39:0,
                 from /usr/local/include/opencv2/flann.hpp:48,
                 from /usr/local/include/opencv2/flann/flann.hpp:48,
                 from /usr/include/pcl-1.8/pcl/kdtree/flann.h:50,
[... 35 lines cut ...]
/usr/include/pcl-1.8/pcl/kdtree/impl/kdtree_flann.hpp:150:37: **error**: ‘Matrix’ is not a member of ‘flann’
   flann_index_->knnSearch (::flann::Matrix<float> (&query[0], 1, dim_),
                                     ^~~~~~
/usr/include/pcl-1.8/pcl/kdtree/impl/kdtree_flann.hpp:150:44: **error**: expected primary-expression before ‘float’
   flann_index_->knnSearch (::flann::Matrix<float> (&query[0], 1, dim_),
                                            ^~~~~
/usr/include/pcl-1.8/pcl/kdtree/impl/kdtree_flann.hpp:151:28: **error**: ‘k_indices_mat’ was not declared in this scope
                            k_indices_mat, k_distances_mat,
                            ^~~~~~~~~~~~~
/usr/include/pcl-1.8/pcl/kdtree/impl/kdtree_flann.hpp:151:28: note: suggested alternative: ‘k_indices’
                            k_indices_mat, k_distances_mat,
                            ^~~~~~~~~~~~~
                            k_indices
/usr/include/pcl-1.8/pcl/kdtree/impl/kdtree_flann.hpp:151:43: error: ‘k_distances_mat’ was not declared in this scope
                            k_indices_mat, k_distances_mat,
                                           ^~~~~~~~~~~~~~~
/usr/include/pcl-1.8/pcl/kdtree/impl/kdtree_flann.hpp:151:43: note: suggested alternative: ‘k_distances’
                            k_indices_mat, k_distances_mat,
                                           ^~~~~~~~~~~~~~~
                                           k_distances
/usr/include/pcl-1.8/pcl/kdtree/impl/kdtree_flann.hpp: In member function ‘int pcl::KdTreeFLANN<PointT, Dist>::radiusSearch(const PointT&, double, std::vector<int>&, std::vector<float>&, unsigned int) const’:
/usr/include/pcl-1.8/pcl/kdtree/impl/kdtree_flann.hpp:184:25: error: ‘params’ has incomplete type
   ::flann::SearchParams params (param_radius_);

`

nathanhhughes, 2022-11-14:
Hi, thanks for your interest in our work (and for reporting this issue)! 

This is something we've encountered on our end on one of our Xaviers due to a conflict between the PCL and OpenCV version that got installed. I'm not actually sure as to what causes the mismatch yet; I suspected Nvidia's custom OpenCV version for a little bit, but I don't remember finding anything conclusive when I was debugging this issue.

The issue isn't limited to Hydra: I've seen that some people have had luck modifying some of their installed header files, but I don't recommend this (which is why I'm not linking to any documentation). We were able solve this issue internally by avoiding including certain PCL headers and opencv headers at the same time in the Hydra code base. It should be pretty feasible to release the fixes as separate branches in [spark_dsg](https://github.com/MIT-SPARK/Spark-DSG) and this repo, hopefully by the end of the week.

nathanhhughes, 2022-11-18:
Hi, please try [this](https://github.com/MIT-SPARK/Spark-DSG/tree/fix/pcl_flann_error) branch in Spark-DSG together with [this](https://github.com/MIT-SPARK/Hydra/tree/fix/pcl_flan_error) branch for Hydra when you get a chance. They *should* work, i.e., they compile on my workstation, but I haven't actually tested them on the Xavier where we've observed the issue internally. I have previously verified that similar changes to our internal version of Hydra did fix the issue on the Xavier, just not these cherry-picked changes.

fengdelin, 2022-11-22:
> Hi, please try [this](https://github.com/MIT-SPARK/Spark-DSG/tree/fix/pcl_flann_error) branch in Spark-DSG together with [this](https://github.com/MIT-SPARK/Hydra/tree/fix/pcl_flan_error) branch for [...]

Thanks,  it can perfectly solve the problem I raised.

closed 2022-11-22 by fengdelin