MIT-SPARK/Hydra #22: [QUESTION] Retrieve absolute positions of the object with respect to the world frame

aswingururaj, 2022-11-17:
Hi,

Thanks a lot for releasing your fantastic work with Hydra. I am working on retrieving object positions from the scene graph that is getting built incrementally. I am testing it on a robot with ORBSLAM2 (for which I had to make some TF modifications) instead of Kimera. However, I noticed that the object nodes I extracted from the scene graph have positions outside the limits of the bounding boxes (min and max parameters). This was not the case in the bag that you had given. Could this be because I modified some TF when I integrated with ORBSLAM2? The pose estimate that I get along with Hydra's reconstruction looks good, and I can extract the objects and see the bounding boxes. Do you have any idea why this could be happening?

Another question I had was regarding the object's position in the world frame. I noticed that even in your bag file, the position I retrieved from the scene graph and the visualization I saw on Rviz were inconsistent. Are the objects being scaled before they're visualized? Then, is the position stored in the scene graph the ground-truth position? 

I am retrieving the object position and the bounding box information at the end of the updateGraph function using the active object's ID. 

Thanks a lot for your help.

nathanhhughes, 2022-11-18:
Thank you for your interest! It's worth noting that convention-wise, the bounding box fields follow pretty exactly what PCL provides. If you're extracting axis-aligned bounding boxes (the default for uhumans2), the min and max will be expressed in the world frame, and if you're extracting oriented bounding boxes (or the z-only oriented bounding boxes) the min and max are expressed slightly differently (see [here](https://pointclouds.org/documentation/classpcl_1_1_moment_of_inertia_estimation.html#a86e92ba5fc831ebbe4c106e1e0c60a8b)). The argument `use_oriented_bounding_boxes` toggles between axis-aligned (if false) and the z-only oriented bounding boxes (if true).  This may be a source of confusion.

What you're seeing might also have to do with synchronization issues between the frontend and the backend (in particular, the backend might not have access to the newest mesh that the frontend provides when updating the scene graph). However, I'm not sure that these issues would manifest in the way you're seeing. 

Can you share more information about what you're seeing? How different is the position from the bounding box? Is it limited to a certain set of object nodes (i.e. the ones in the active window), or is it all of them? Saving the scene graph (at the point where you're seeing the issue) and then plotting the resulting objects and bounding boxes may be a good way to look at this in a more principled manner.

> Are the objects being scaled before visualized, or do bounding boxes have some other reference coordinate frame? Then, is the position stored in the scene graph the ground-truth position?

No, the objects and bounding box are being visualized as is, i.e. the bounding boxes are CUBE markers drawn around `attributes.bounding_box.world_P_center` with scale `attributes.bounding_box.max - attributes.bounding_box.min`, and the node markers are drawn around `attributes.position`. Both use the current estimates (we don't have access to ground truth object information inside Hydra).

aswingururaj, 2022-11-27:
Apologies for my delayed reply. Thanks for your prompt response. I checked again, and it looks like I was visualizing it wrong. Thanks for the heads up on using oriented bounding boxes; it did help me to fix the issue. Thanks again!

nathanhhughes, 2022-12-09:
Great, glad that you found the issue! I'm going to close this for now, feel free to reopen if anything related pops up

closed 2022-12-09 by nathanhhughes