MIT-SPARK/Hydra #25: Errors in Kimera-RPGO while building via catkin

KirillMouraviev, 2023-01-26:
Hi,
I tried to build Hydra as it written in the Readme, and, while running `catkin build`, faced with the following compilation errors:

[image]
[image]

Addition of `set(CMAKE_CXX_STANDARD 17)` into `CMakeLists.txt` files in GTSAM and Kimera-RPGO did not help.

My system configuration: OS Ubuntu 20.04 LTS, ROS Noetic, GCC compiler version 9.4.0.

Could you say, which is the best way to fix these CMake errors?

nathanhhughes, 2023-01-26:
Hi, thanks for reporting this issue! GTSAM recently (in the past few days) started removing usage of various `boost` classes and replacing them with their standard library versions. We're working on updating Hydra and various dependencies (Kimera-RPGO and Kimera-PGMO) to match, but it will be a while before we finish. You should be able to use version `4.2a8` for GTSAM in the meantime (`git checkout 4.2a8` in gtsam's directory).  I'll also push changes to the rosinstall file when I get a chance next week.

KirillMouraviev, 2023-01-27:
With version `4.2a8`, it compiled. Thank you!

closed 2023-01-27 by KirillMouraviev