MIT-SPARK/Hydra #26: Compilation Errors

jerryepratt, 2023-02-10:
I am trying to compile Hydra on Ubuntu 20.04 with ROS Noetic, using the instructions at https://github.com/MIT-SPARK/Hydra using catkin.

I had a compilation problem for Kimera-RPGO, which was reported and fix suggested at https://github.com/MIT-SPARK/Kimera-RPGO/issues/86. I implemented that fix by changing the Kimera-RPGO version to be fix/gtsam_boost_removal in /root/catkin_ws/src/hydra/install/hydr.rosinstall before running
vcs import . < hydra/install/hydra.rosinstall

That fixed the Kimera-RPGO error, but now I get the following error:

Errors     << kimera_pgmo:make /root/catkin_ws/logs/kimera_pgmo/build.make.000.log                                                              
In file included from /root/catkin_ws/devel/include/gtsam/base/Matrix.h:28,
                 from /root/catkin_ws/devel/include/gtsam/base/Manifold.h:22,
                 from /root/catkin_ws/devel/include/gtsam/geometry/BearingRange.h:21,
                 from /root/catkin_ws/devel/include/gtsam/geometry/Pose3.h:22,
                 from /root/catkin_ws/src/kimera_pgmo/include/kimera_pgmo/utils/CommonFunctions.h:12,
                 from /root/catkin_ws/src/kimera_pgmo/src/compression/MeshCompression.cpp:11:
/root/catkin_ws/devel/include/gtsam/base/Vector.h:75:52: error: static assertion failed: Error: GTSAM was built against a different version of Eigen
   75 |     GTSAM_EIGEN_VERSION_WORLD==EIGEN_WORLD_VERSION &&
      |                                                    ^

jerryepratt, 2023-02-10:
(Sorry, this is just a compilation issue, not a bug report. Didn't mean to add the bug label...)

nathanhhughes, 2023-02-11:
Hi, thanks for reporting this (and no worries about the bug label, it's added automatically)! I just pushed the change to the rosinstall file that I meant to make last week in response to #25 (which stems from the same problem you're having) to pin GTSAM to a specific version temporarily (`4.2a9`). My recommendation is to use one of the gtsam 4.2 pre-release tags; I've been able to build against `4.2a8` and `4.2a9` without problems on a newer version of Hydra, but haven't actually tested building `main` against `4.2a9`.

Side note: the branch that Yun pointed you to in Kimera-RPGO will still work with GTSAM version `4.2a9`, but I'd actually recommend sticking with Kimera-RPGO `master` for now (sorry, Yun and I should have coordinated better).

jerryepratt, 2023-02-14:
Thanks! Your change fixed the issue and everything is building now using the instructions at https://github.com/MIT-SPARK/Hydra

closed 2023-02-14 by jerryepratt

LongfeiDE, 2023-05-23:
hello, i have the same error complaining about GTSAM was built against a different version of Eigen when building hydra_utils triggered by pose_graph_publisher_node.cpp:37
I have -DGTSAM_USE_SYSTEM_EIGEN=ON which helps with some other packages but this one cannot pass.