MIT-SPARK/Hydra #27: [QUESTION] ROS topic for the scene graph

clams-casino, 2023-02-13:
Hi,

I'm wondering if there is a ROS topic that publishes the current scene graph?

Thank you!

nathanhhughes, 2023-02-13:
Hi, thanks for your interest in our work! By default, the DSG is published on the topic `/incremental_dsg_builder_node/dsg` (the visualizer launch file has the topic [here](https://github.com/MIT-SPARK/Hydra/blob/ad70db03f7fa736a37fcbd63e9e4e928e4028ec0/hydra_utils/launch/hydra_streaming_visualizer.launch#L3) as an argument).  You can see an example of how to access the received scene graph [here](https://github.com/MIT-SPARK/Hydra/blob/ad70db03f7fa736a37fcbd63e9e4e928e4028ec0/hydra_utils/src/hydra_visualizer_node.cpp#L187-L196) (we wrap the actual ros publisher with some code to handle reconstructing the scene graph from the ros message that we use to pass it around).

clams-casino, 2023-02-13:
Thank you for the quick response!

closed 2023-02-13 by clams-casino