MIT-SPARK/Hydra #28: [QUESTION] Support for multi-session and multi-agent

araujokth, 2023-02-16:
Hi, I was wondering if you have support for multi-session (e.g. save the 3D scene graph after a session, and re-load it later to keep building on it) and also multi-agent (e.g. two agents can be running at the same time, contributing to the same 3D scene graph)? I ran the code and read through the code and paper and could not find information on that. If not, is it something that you plan/do not plan to support? Thank you!

nathanhhughes, 2023-03-08:
Hi, thank you for your interest in our work! We do not support or plan to support multi-session mapping in the near-term (stay tuned for a paper on the multi-agent side though).

araujokth, 2023-03-09:
Thanks for the reply Nathan! Very looking forward for the paper on the multi-agent problem and it would be great to try it out in the future if you release the code!

closed 2023-09-19 by nathanhhughes