MIT-SPARK/Hydra #29: missing config file of the scene "neighborhood1" in the folder "hydra_dsg_builder/config/uhumans2/" ?

JimPlayboy, 2023-02-19:
Hi Nathan

We met the same question as issue #18, and have successfully solved the problem with the following command:
`roslaunch hydra_dsg_builder uhumans2_incremental_dsg.launch start_visualizer:=true semantic_map_file:=tesse_multiscene_archviz1_segmentation_mapping.csv typology_file:=uhmans2_apartment_typology.yaml
`

It works well for office, archviz(apartment) and underground(subway), but it seems that the corresponding typology file of "tesse_multiscene_neighborhood1_segmentation_mapping.csv" is missing in the folder "hydra_dsg_builder/config/uhumans2/".

Besides, the mesh of neighborhood seems to be limited to 2d ground plane. We wonder if it works well?  

Can you help us with that?
Thanks!

nathanhhughes, 2023-03-08:
Hi, thank you for your interest in our work! We didn't test against the neighborhood scene when developing Hydra and writing our RSS paper for the reason you mention: it's an outdoor scene with mostly 2D geometry, and the focus of Hydra has been on indoor scenes. 

That said, you can just write a typology file for the scene if you do want to test it. You would only have to fill out the fields for dynamic labels and object labels (the other fields were used by Kimera, but we no longer use them). The dynamic label (20) should stay the same, and the object labels field is just populated with the semantic labels you want to extract objects for.

nathanhhughes, 2023-09-19:
I'm going to close this issue. Feel free to re-open if you have further questions!