MIT-SPARK/Hydra #30: [QUESTION] Release visual loop-closures?

huma0605, 2023-02-20:
Hi, one of the critical contribution of this work is about the loop-closures, but the current release is missing the visual loop-closure part. May I ask do you plan to release it so that other works could benchmark the real performance of this work?

Atellas23, 2023-02-23:
Hey, I would like to ask for this as well!

araujokth, 2023-02-27:
Hi, you can refer to https://github.com/MIT-SPARK/Hydra/issues/1 for further info on the loop closures from @nathanhhughes. He also mentioned they were trying to release the code for this before the IROS deadline which is this week

nathanhhughes, 2023-03-08:
@huma0605 @Atellas23 Thank you for your interest!  We've made a pre-release of Kimera-VIO and Kimera-VIO-ROS available with the required interface to publish visual loop closure information. Please refer [here](https://github.com/MIT-SPARK/Hydra#using-kimera-vio) for instructions.

closed 2023-03-08 by nathanhhughes