MIT-SPARK/Hydra #31: [QUESTION] How to enable dynamic nodes visualization?

MikeDmytrenko, 2023-03-06:
Hi @nathanhhughes! Is there a way to track spatial information for other dynamic objects as was shown in the Kimera paper? For example, how can I track human trajectories in the uHumans dataset? In the resultant dsg.json, I can only see nodes with trajectory related to the agent mapping the scene.

Thank you so much for your attention and participation.

nathanhhughes, 2023-03-08:
Hi, thanks for the question! We did not implement the functionality for human pose-graph tracking in Hydra that the various Kimera DSG papers had. This is something that we're thinking about looking into, but no there's no definite timeline on that. 

That said, the visualizer will show the nodes for any dynamic layer added to the graph (so if someone added human pose-graph tracking to Hydra, the visualizer should just work).

nathanhhughes, 2023-09-19:
Closing this for now. If you have any more questions w.r.t. how the visualization of dynamic layers works, feel free to reopen

closed 2023-09-19 by nathanhhughes