MIT-SPARK/Hydra #33: [QUESTION] Semantic color and label mapping in uHumans2 office dataset

araujokth, 2023-06-03:
Hi! I was looking a bit closer into the semantic image from the office rosbag and what labels come out from Hydra, and it looks like the label and color mapping is not corresponding to the file I thought was creating this mapping.

My understanding was that the color to label mapping was set by this file https://github.com/MIT-SPARK/Kimera-Semantics/blob/master/kimera_semantics_ros/cfg/tesse_multiscene_office1_segmentation_mapping.csv as its the one defined in the launch file for uhumans2_incremental_dsg, but when I check the csv colors and respective labels, only chairs and couches seem to have the correct assignment (green -> 5). Other objects like LCD (should be red -> 10 in csv, but its blue in the semantic image), plants (should be blue -> 13 in csv, but its orange in the semantic image), water dispenser (should be dark red  -> 18 in csv, but its purple in the semantic image). See snapshot below. However, all the id numbers seem to be correct. Is there some other remapping happening, am I looking at the wrong csv file, or am I confusing something? :) 

Thanks for the help!

[image]

nathanhhughes, 2023-06-03:
Hi, thanks for your interest in our work! I suspect that you have a BGR vs. RGB ordering issue. Not sure whether it's how you're accessing or displaying the images, but just check the image message encoding field to make sure you have the right starting convention.

araujokth, 2023-06-03:
Thanks Nathan! That was it.. sorry for the silly question and thanks again!

closed 2023-06-03 by araujokth