MIT-SPARK/Hydra #34: [QUESTION] problems with angular movement

michaelDengl, 2023-06-03:
Hi Nathan,

We have corresponded with you once or twice over the past couple of months. We are now in the final stages of implementing Hydra with our own equipment in our laboratory in Zürich. Everything is coming along well, and we're currently addressing the last few bugs we've encountered. There is one bug, however, that we're having trouble understanding. It first appeared during the simulation stage in Gazebo, and is still occurring now as we operate the robot in real-world conditions. This bug causes the robot's position to become entirely skewed during angular movement. A simple rotation is depicted on the graph as a large circle, giving the impression that the robot has covered a significant distance, as illustrated in the picture posted below. Do you have any insights into why this could be happening? Please bear in mind that we are currently not using Kimera-VIO, but are solely depending on the tf topics.

Thank you very much in advance for your assistance.

Kind regards,
Michael Dengl & Lilla Goldmann

[image]

nathanhhughes, 2023-07-26:
If I'm understanding it correctly, you're using ground-truth poses? If that's the case, then it's likely that you have your tf tree set up incorrectly and it's a matter of making sure that the specified tf frame that Hydra is using is the camera's frame.

If you're not using ground-truth poses, it's worth noting that many VIO / odometry algorithms struggle with rotation only movements and drift while rotating in place is quite common.

michaelDengl, 2023-07-27:
Thanks for the feedback, it was indeed a problem with the quaternion calculcation and the tf tree. We finished our thesis by now, it was a very intressting journey. If you're curious how we did it from a-z I can send you our thesis by mail. Just let me know.

closed 2023-09-19 by nathanhhughes

nvhungv2k, 2024-08-05:
> Thanks for the feedback, it was indeed a problem with the quaternion calculcation and the tf tree. We finished our thesis by now, it was a very intressting journey. If you're curious how we did it from a-z I can send you our thesis by mail. Just let me know.

Hi @michaelDengl 
Can you share how to create hydra-based DSG from Gazebo environment?

michaelDengl, 2024-08-09:
Hey Nguyễn Hùng, 

we needed 4 months for this as part of our bachelor thesis. This is not easyly explained Im afraid.

Kind regards
Michael

nvhungv2k, 2024-08-09:
> Hey Nguyễn Hùng,
> 
> we needed 4 months for this as part of our bachelor thesis. This is not easyly explained Im afraid.
> 
> Kind regards Michael

Thanks for your reply!
I know that it is not easy to explain. So could you please share me privately via my email: nvhung.v2k@gmail.com?
Thanks a lot!

michaelDengl, 2024-08-13:
I can also tell you here: we did it with hundrets of hours of work and a lot of trial and error. This was as mentioned part of our bachelor thesis. I have no longer access to the code nor the environment as this implementation was done on a kubernetes cluster to which I no longer have access to.