MIT-SPARK/Hydra #42: Error: GTSAM was built against a different version of Eigen

skarthikbe96, 2023-11-03:
Hello I am trying to do catkin build the Hydra package,
 it throws me an error while building the hydra_ros package

error: static assertion failed: Error: GTSAM was built against a different version of Eigen
   75 |     GTSAM_EIGEN_VERSION_WORLD==EIGEN_WORLD_VERSION &&

I tried https://github.com/MIT-SPARK/Kimera-VIO-ROS/issues/12 fixing the issue but it didn't help
I am using ros1 and ubuntu 20.04

nathanhhughes, 2023-11-07:
Thanks for your interest in our work! Please make sure you are using the correct version of GTSAM (`4.2a8` is the current version that the rosinstall file for Hydra specifies). That said, specifying `-DGTSAM_USE_SYSTEM_EIGEN=ON` (and rebuilding) should also fix the issue.

nathanhhughes, 2024-07-09:
Note that Hydra now uses the version of GTSAM vendored by ROS (see the package announcement [here]( https://discourse.ros.org/t/announcing-gtsam-as-a-ros-1-2-package/32739)), so this no longer applies

closed 2024-07-09 by nathanhhughes