MIT-SPARK/Hydra #53: Regarding VIO

skarthikbe96, 2024-07-09:
Thanks for your great work, as hydra takes kimera-vio as odometry input. I am currently working on monocular camera odometry inputs and want to use other odometry input like vins-fusion, orb-slam3. What are the information does it required for the odometry to make hydra run. Thank you

nathanhhughes, 2024-08-05:
Hi, thanks for your interest in our work! Hydra only requires the pose (which it looks up via TF2) from an odometry source. In addition, Hydra can either be configured to subscribe to a  semantically labeled pointcloud (so depth + semantics and optionally color) as an input or rgb, depth and semantic label images directly (instead of the labeled pointcloud). Feel free to take a look at the launch files in [Hydra-Ros](https://github.com/MIT-SPARK/Hydra-ROS/tree/main/hydra_ros/launch) for more details.

As a further note, it might be helpful to point out that if you're using monocular VIO as odometry, your pose will have scale drift and you won't be able to get metric depth out of the camera. While I'm aware of at least one case where this was done, it's not really supported in Hydra (most crucially, our backend does not optimize for scale drift) and I don't really plan on supporting it. Good luck on your attempt!

closed 2024-08-05 by nathanhhughes

skarthikbe96, 2024-08-06:
Thank you for your valuable reply. It helps me a lot i manage to ran monocular VIO and it produce good results. Do you have any suggestion for loop closue instead of using kimera-vio lcd that could produce this [https://github.com/MIT-SPARK/pose_graph_tools/blob/feature/hydra/msg/BowQuery.msg](message) to do loop closure