MIT-SPARK/Hydra #54: Initialting Shutdown.

Hugh-Chen-Hugues, 2024-07-14:
Hi, thx for your wonderful work. I've compiled your most recent Hydra project successfully. But when I follow your tutorial typing `roslaunch hydra_ros uhumans2.launch
` 
in the terminal. It turns out to be failling at initializing. Below are the informations reported from terminal. `roslaunch hydra_ros uhumans2.launch
... logging to /home/hugh/.ros/log/c22967fc-4191-11ef-bdd4-2ba20d364c2b/roslaunch-hugh-ZHENGJIUZHE-REN9000-25ICZ-35219.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://hugh-ZHENGJIUZHE-REN9000-25ICZ:42483/

SUMMARY
========

PARAMETERS
 * /hydra_dsg_visualizer/config/collapse_layers: False
 * /hydra_dsg_visualizer/config/color_places_by_distance: False
 * /hydra_dsg_visualizer/config/dynamic_layer/2/color_offset: 3
 * /hydra_dsg_visualizer/config/dynamic_layer/2/edge_alpha: 0.9
 * /hydra_dsg_visualizer/config/dynamic_layer/2/edge_scale: 0.05
 * /hydra_dsg_visualizer/config/dynamic_layer/2/edge_sl_ratio: 0.6
 * /hydra_dsg_visualizer/config/dynamic_layer/2/interlayer_edge_insertion_skip: 10
 * /hydra_dsg_visualizer/config/dynamic_layer/2/label_height: 0.9
 * /hydra_dsg_visualizer/config/dynamic_layer/2/label_scale: 0.8
 * /hydra_dsg_visualizer/config/dynamic_layer/2/luminance: 0.7
 * /hydra_dsg_visualizer/config/dynamic_layer/2/node_alpha: 0.9
 * /hydra_dsg_visualizer/config/dynamic_layer/2/node_scale: 0.2
 * /hydra_dsg_visualizer/config/dynamic_layer/2/node_use_sphere: True
 * /hydra_dsg_visualizer/config/dynamic_layer/2/num_colors: 5
 * /hydra_dsg_visualizer/config/dynamic_layer/2/saturation: 0.9
 * /hydra_dsg_visualizer/config/dynamic_layer/2/visualize: True
 * /hydra_dsg_visualizer/config/dynamic_layer/2/visualize_interlayer_edges: False
 * /hydra_dsg_visualizer/config/dynamic_layer/2/z_offset_scale: 0.0
 * /hydra_dsg_visualizer/config/layer2/bbox_wireframe_edge_scale: 0.05
 * /hydra_dsg_visualizer/config/layer2/bbox_wireframe_scale: 0.05
 * /hydra_dsg_visualizer/config/layer2/bounding_box_alpha: 0.9
 * /hydra_dsg_visualizer/config/layer2/collapse_bounding_box: True
 * /hydra_dsg_visualizer/config/layer2/interlayer_edge_alpha: 0.3
 * /hydra_dsg_visualizer/config/layer2/interlayer_edge_insertion_skip: 10
 * /hydra_dsg_visualizer/config/layer2/interlayer_edge_scale: 0.05
 * /hydra_dsg_visualizer/config/layer2/interlayer_edge_use_color: True
 * /hydra_dsg_visualizer/config/layer2/intralayer_edge_alpha: 0.6
 * /hydra_dsg_visualizer/config/layer2/intralayer_edge_insertion_skip: 0
 * /hydra_dsg_visualizer/config/layer2/intralayer_edge_scale: 0.03
 * /hydra_dsg_visualizer/config/layer2/intralayer_edge_use_color: False
 * /hydra_dsg_visualizer/config/layer2/label_height: 0.5
 * /hydra_dsg_visualizer/config/layer2/label_scale: 0.45
 * /hydra_dsg_visualizer/config/layer2/marker_alpha: 0.6
 * /hydra_dsg_visualizer/config/layer2/marker_color_mode: 0
 * /hydra_dsg_visualizer/config/layer2/marker_scale: 0.4
 * /hydra_dsg_visualizer/config/layer2/use_bounding_box: True
 * /hydra_dsg_visualizer/config/layer2/use_collapsed_label: True
 * /hydra_dsg_visualizer/config/layer2/use_edge_source: True
 * /hydra_dsg_visualizer/config/layer2/use_label: False
 * /hydra_dsg_visualizer/config/layer2/use_sphere_marker: False
 * /hydra_dsg_visualizer/config/layer2/visualize: True
 * /hydra_dsg_visualizer/config/layer2/z_offset_scale: 2.0
 * /hydra_dsg_visualizer/config/layer20/boundary_alpha: 1.0
 * /hydra_dsg_visualizer/config/layer20/boundary_use_node_color: True
 * /hydra_dsg_visualizer/config/layer20/boundary_wireframe_scale: 0.1
 * /hydra_dsg_visualizer/config/layer20/bounding_box_alpha: 0.5
 * /hydra_dsg_visualizer/config/layer20/collapse_boundary: False
 * /hydra_dsg_visualizer/config/layer20/collapse_bounding_box: False
 * /hydra_dsg_visualizer/config/layer20/draw_boundaries: True
 * /hydra_dsg_visualizer/config/layer20/interlayer_edge_alpha: 0.4
 * /hydra_dsg_visualizer/config/layer20/interlayer_edge_insertion_skip: 0
 * /hydra_dsg_visualizer/config/layer20/interlayer_edge_scale: 0.08
 * /hydra_dsg_visualizer/config/layer20/interlayer_edge_use_color: True
 * /hydra_dsg_visualizer/config/layer20/intralayer_edge_alpha: 0.5
 * /hydra_dsg_visualizer/config/layer20/intralayer_edge_insertion_skip: 0
 * /hydra_dsg_visualizer/config/layer20/intralayer_edge_scale: 0.01
[... 410 lines cut ...]
    @     0x55c3af1301bf main
    @     0x7f12a89f5083 __libc_start_main
    @     0x55c3af130cae _start
    @                0x0 (unknown)
================================================================================REQUIRED process [hydra_ros_node-9] has died!
process has died [pid 35308, exit code -11, cmd /home/hugh/catkin_ws/devel/lib/hydra_ros/hydra_ros_node --minloglevel=0 -v=0 ~pose_graph:=kimera_vio_ros/pose_graph_incremental ~agent_node_measurements:=kimera_distributed/optimized_nodes ~bow_vectors:=kimera_vio_ros/bow_query frame_registration:=kimera_vio_ros/kimera_vio_ros_node/register_lcd_frames ~pointcloud:=pointcloud ~rgb/image_raw:=/tesse/left_cam/rgb/image_raw ~rgb/image_raw/compressed:=/tesse/left_cam/rgb/image_raw/compressed ~rgb/camera_info:=/tesse/left_cam/camera_info ~depth_registered/image_rect:=/tesse/depth_cam/mono/image_raw ~semantic/image_raw:=/tesse/seg_cam/rgb/image_raw __name:=hydra_ros_node __log:=/home/hugh/.ros/log/c22967fc-4191-11ef-bdd4-2ba20d364c2b/hydra_ros_node-9.log].
log file: /home/hugh/.ros/log/c22967fc-4191-11ef-bdd4-2ba20d364c2b/hydra_ros_node-9*.log
Initiating shutdown!
================================================================================
[hydra_dsg_visualizer-11] killing on exit
[rviz-10] killing on exit
[hydra_ros_node-9] killing on exit
[fake_world_tf-8] killing on exit
[bag_static_tf_5-7] killing on exit
[bag_static_tf_3-5] killing on exit
[bag_static_tf_2-4] killing on exit
[bag_static_tf_1-3] killing on exit
[bag_static_tf_4-6] killing on exit
[bag_static_tf_0-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done`
Could you please tell me what the problem is? BTW, I've also compiled Kimera-VIO in the catkin_ws, and when I roslaunch kimera-vio, there is no bug, but after that when I tried to roslaunch hydra_ros, it reported the same problem as above.

Hugh-Chen-Hugues, 2024-07-19:
Hi, I rebuild the whole project and now it still can not work, could you please tell me what the problem is? Thx!
```
roslaunch hydra_ros uhumans2.launch
... logging to /home/hugh/.ros/log/815d2696-4583-11ef-ab5f-fd2e822235bf/roslaunch-hugh-ZHENGJIUZHE-REN9000-25ICZ-11069.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://hugh-ZHENGJIUZHE-REN9000-25ICZ:45841/

SUMMARY
========

PARAMETERS
 * /hydra_dsg_visualizer/config/collapse_layers: False
 * /hydra_dsg_visualizer/config/color_places_by_distance: False
 * /hydra_dsg_visualizer/config/dynamic_layer/2/color_offset: 3
 * /hydra_dsg_visualizer/config/dynamic_layer/2/edge_alpha: 0.9
 * /hydra_dsg_visualizer/config/dynamic_layer/2/edge_scale: 0.05
 * /hydra_dsg_visualizer/config/dynamic_layer/2/edge_sl_ratio: 0.6
 * /hydra_dsg_visualizer/config/dynamic_layer/2/interlayer_edge_insertion_skip: 10
 * /hydra_dsg_visualizer/config/dynamic_layer/2/label_height: 0.9
 * /hydra_dsg_visualizer/config/dynamic_layer/2/label_scale: 0.8
 * /hydra_dsg_visualizer/config/dynamic_layer/2/luminance: 0.7
 * /hydra_dsg_visualizer/config/dynamic_layer/2/node_alpha: 0.9
 * /hydra_dsg_visualizer/config/dynamic_layer/2/node_scale: 0.2
 * /hydra_dsg_visualizer/config/dynamic_layer/2/node_use_sphere: True
 * /hydra_dsg_visualizer/config/dynamic_layer/2/num_colors: 5
 * /hydra_dsg_visualizer/config/dynamic_layer/2/saturation: 0.9
 * /hydra_dsg_visualizer/config/dynamic_layer/2/visualize: True
 * /hydra_dsg_visualizer/config/dynamic_layer/2/visualize_interlayer_edges: False
 * /hydra_dsg_visualizer/config/dynamic_layer/2/z_offset_scale: 0.0
 * /hydra_dsg_visualizer/config/layer2/bbox_wireframe_edge_scale: 0.05
 * /hydra_dsg_visualizer/config/layer2/bbox_wireframe_scale: 0.05
 * /hydra_dsg_visualizer/config/layer2/bounding_box_alpha: 0.9
 * /hydra_dsg_visualizer/config/layer2/collapse_bounding_box: True
 * /hydra_dsg_visualizer/config/layer2/interlayer_edge_alpha: 0.3
 * /hydra_dsg_visualizer/config/layer2/interlayer_edge_insertion_skip: 10
 * /hydra_dsg_visualizer/config/layer2/interlayer_edge_scale: 0.05
 * /hydra_dsg_visualizer/config/layer2/interlayer_edge_use_color: True
 * /hydra_dsg_visualizer/config/layer2/intralayer_edge_alpha: 0.6
 * /hydra_dsg_visualizer/config/layer2/intralayer_edge_insertion_skip: 0
 * /hydra_dsg_visualizer/config/layer2/intralayer_edge_scale: 0.03
 * /hydra_dsg_visualizer/config/layer2/intralayer_edge_use_color: False
 * /hydra_dsg_visualizer/config/layer2/label_height: 0.5
 * /hydra_dsg_visualizer/config/layer2/label_scale: 0.45
 * /hydra_dsg_visualizer/config/layer2/marker_alpha: 0.6
 * /hydra_dsg_visualizer/config/layer2/marker_color_mode: 0
 * /hydra_dsg_visualizer/config/layer2/marker_scale: 0.4
 * /hydra_dsg_visualizer/config/layer2/use_bounding_box: True
 * /hydra_dsg_visualizer/config/layer2/use_collapsed_label: True
 * /hydra_dsg_visualizer/config/layer2/use_edge_source: True
 * /hydra_dsg_visualizer/config/layer2/use_label: False
 * /hydra_dsg_visualizer/config/layer2/use_sphere_marker: False
 * /hydra_dsg_visualizer/config/layer2/visualize: True
 * /hydra_dsg_visualizer/config/layer2/z_offset_scale: 2.0
 * /hydra_dsg_visualizer/config/layer20/boundary_alpha: 1.0
 * /hydra_dsg_visualizer/config/layer20/boundary_use_node_color: True
 * /hydra_dsg_visualizer/config/layer20/boundary_wireframe_scale: 0.1
 * /hydra_dsg_visualizer/config/layer20/bounding_box_alpha: 0.5
 * /hydra_dsg_visualizer/config/layer20/collapse_boundary: False
 * /hydra_dsg_visualizer/config/layer20/collapse_bounding_box: False
 * /hydra_dsg_visualizer/config/layer20/draw_boundaries: True
 * /hydra_dsg_visualizer/config/layer20/interlayer_edge_alpha: 0.4
 * /hydra_dsg_visualizer/config/layer20/interlayer_edge_insertion_skip: 0
 * /hydra_dsg_visualizer/config/layer20/interlayer_edge_scale: 0.08
 * /hydra_dsg_visualizer/config/layer20/interlayer_edge_use_color: True
 * /hydra_dsg_visualizer/config/layer20/intralayer_edge_alpha: 0.5
 * /hydra_dsg_visualizer/config/layer20/intralayer_edge_insertion_skip: 0
 * /hydra_dsg_visualizer/config/layer20/intralayer_edge_scale: 0.01
[... 418 lines cut ...]
    @     0x5607fe5fd5ab main
    @     0x7f7085de7083 __libc_start_main
    @     0x5607fe5fd6be _start
    @                0x0 (unknown)
================================================================================REQUIRED process [hydra_ros_node-9] has died!
process has died [pid 11161, exit code -11, cmd /home/hugh/catkin_ws/devel/lib/hydra_ros/hydra_ros_node --minloglevel=0 -v=0 ~pose_graph:=kimera_vio_ros/pose_graph_incremental ~agent_node_measurements:=kimera_distributed/optimized_nodes ~bow_vectors:=kimera_vio_ros/bow_query frame_registration:=kimera_vio_ros/kimera_vio_ros_node/register_lcd_frames ~pointcloud:=pointcloud ~rgb/image_raw:=/tesse/left_cam/rgb/image_raw ~rgb/image_raw/compressed:=/tesse/left_cam/rgb/image_raw/compressed ~rgb/camera_info:=/tesse/left_cam/camera_info ~depth_registered/image_rect:=/tesse/depth_cam/mono/image_raw ~semantic/image_raw:=/tesse/seg_cam/rgb/image_raw __name:=hydra_ros_node __log:=/home/hugh/.ros/log/815d2696-4583-11ef-ab5f-fd2e822235bf/hydra_ros_node-9.log].
log file: /home/hugh/.ros/log/815d2696-4583-11ef-ab5f-fd2e822235bf/hydra_ros_node-9*.log
Initiating shutdown!
================================================================================
[hydra_dsg_visualizer-11] killing on exit
[rviz-10] killing on exit
[fake_world_tf-8] killing on exit
[bag_static_tf_4-6] killing on exit
[bag_static_tf_5-7] killing on exit
[bag_static_tf_2-4] killing on exit
[bag_static_tf_3-5] killing on exit
[bag_static_tf_1-3] killing on exit
[bag_static_tf_0-2] killing on exit
[hydra_ros_node-9] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

closed 2024-07-24 by Hugh-Chen-Hugues

warlrdBarton, 2024-08-22:
Did you solve the problem? I encounter it as well.

Hugh-Chen-Hugues, 2024-09-01:
> Did you solve the problem? I encounter it as well.

I reinstalled my Ubuntu system and then built hydra in the clean ubuntu system, and it worked.

warlrdBarton, 2024-09-03:
Thx for ur reply. I also found the problem lies in catkin build calls a
wrong gtsam in ros built-in directory, instead of the one we installed in
system. I uninstalled the one in ros directory, and specified explicitly in
cmakelists. then I can compile it successfully.