MIT-SPARK/Hydra #55: [QUESTION] Discrepancy Between Semantic Segmentation Colors and Mesh Colors

B-Paweekorn, 2024-07-17:
I'm very interested in your work. I think I'm in the final stages of my project, but I've encountered a challenge. The mesh colors displayed in RViz do not match the semantic colors provided in the input. This issue is evident in the attached screenshot, where the semantic segmentation colors differ from the mesh colors.

Could you please advise on potential reasons for this discrepancy?

B-Paweekorn, 2024-07-17:
[image]

nathanhhughes, 2024-08-05:
Hi, thanks for your interest in our work! It's hard to say without more details, but there are two spots where we've managed to have misaligned colors:
  -  The visualizer and hydra can have different settings for the `semantic_map_path` argument (which doesn't look like it is the case as the object color matches the mesh color). 
  - The image encoding might be flipped from what you expect at the source (i.e., you could be publishing a `rgb` ordered image with a `bgr` ordered encoding or vice-versa). 
 
Finally, there's also a possibility you might not have set `semantic_map_path` correctly to point to the current colormap that you're using