MIT-SPARK/Hydra #56: [BUG] Error Building catkin

Optimax14, 2024-07-17:
While following the installing instructions on the main repo I got a building error while using catkin build
the log is in the attached txt file
[build.txt](https://github.com/user-attachments/files/16274620/build.txt)

additionally when running rosdep install --from-paths . --ignore-src -r -y i was faced with an error
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
kimera_pgmo_ros: Cannot locate rosdep definition for [voxblox_msgs]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully

Please help... Thank you

Hugh-Chen-Hugues, 2024-07-18:
A possible solution maybe:
```
cd path/to/catkin_ws/src
git clone https://github.com/MIT-SPARK/voxblox.git
cd ..
catkin build
```
I am not sure this could solve your problem, but you can give it a try.

nathanhhughes, 2024-08-05:
Hi, thanks for your interest in our work! Unfortunately, the response from @Hugh-Chen-Hugues is incorrect and adding voxblox to the workspace is not going to fix any issues. In this case, `voxblox_msgs` is an optional dependency for `Kimera-PGMO` and the rosdep error can be ignored.  

Also, the included build log looks like it only captured stdout and not stderr and I can't tell what the actual error is