MIT-SPARK/Hydra #59: [BUG] python bindings

niyatirawal, 2024-07-30:
For installing the python package, when I do the last ``` pip install .``` , I get the following error. Any idea on how to fix it?
```       
      -- Build files have been written to: /home/lbaraldi/catkin_ws/src/hydra/build/temp.linux-x86_64-cpython-38
      [  1%] Building CXX object CMakeFiles/hydra.dir/src/backend/backend_module.cpp.o
      [  2%] Building CXX object CMakeFiles/hydra.dir/src/backend/backend_utilities.cpp.o
      [  3%] Building CXX object CMakeFiles/hydra.dir/src/backend/pgmo_configs.cpp.o
      [  4%] Building CXX object CMakeFiles/hydra.dir/src/backend/merge_tracker.cpp.o
      [  5%] Building CXX object CMakeFiles/hydra.dir/src/backend/surface_place_utilities.cpp.o
      [  6%] Building CXX object CMakeFiles/hydra.dir/src/backend/update_agents_functor.cpp.o
      [  7%] Building CXX object CMakeFiles/hydra.dir/src/backend/update_frontiers_functor.cpp.o
      [  8%] Building CXX object CMakeFiles/hydra.dir/src/backend/update_functions.cpp.o
      [  8%] Building CXX object CMakeFiles/hydra.dir/src/backend/update_objects_functor.cpp.o
      [  9%] Building CXX object CMakeFiles/hydra.dir/src/backend/update_places_functor.cpp.o
      [ 10%] Building CXX object CMakeFiles/hydra.dir/src/backend/update_rooms_buildings_functor.cpp.o
      [ 11%] Building CXX object CMakeFiles/hydra.dir/src/backend/update_surface_places_functor.cpp.o
      [ 12%] Building CXX object CMakeFiles/hydra.dir/src/common/batch_pipeline.cpp.o
      [ 13%] Building CXX object CMakeFiles/hydra.dir/src/common/config_utilities.cpp.o
      [ 14%] Building CXX object CMakeFiles/hydra.dir/src/common/global_info.cpp.o
      [ 15%] Building CXX object CMakeFiles/hydra.dir/src/common/hydra_pipeline.cpp.o
      [ 16%] Building CXX object CMakeFiles/hydra.dir/src/common/label_remapper.cpp.o
      [ 17%] Building CXX object CMakeFiles/hydra.dir/src/common/label_space_config.cpp.o
      [ 18%] Building CXX object CMakeFiles/hydra.dir/src/common/robot_prefix_config.cpp.o
      [ 19%] Building CXX object CMakeFiles/hydra.dir/src/common/semantic_color_map.cpp.o
      [ 20%] Building CXX object CMakeFiles/hydra.dir/src/common/shared_dsg_info.cpp.o
      [ 21%] Building CXX object CMakeFiles/hydra.dir/src/common/shared_module_state.cpp.o
      [ 22%] Building CXX object CMakeFiles/hydra.dir/src/frontend/frontend_module.cpp.o
      [ 23%] Building CXX object CMakeFiles/hydra.dir/src/frontend/gvd_place_extractor.cpp.o
      [ 24%] Building CXX object CMakeFiles/hydra.dir/src/frontend/frontier_extractor.cpp.o
      [ 25%] Building CXX object CMakeFiles/hydra.dir/src/frontend/mesh_segmenter.cpp.o
      c++: fatal error: Killed signal terminated program cc1plus
      compilation terminated.
      make[2]: *** [CMakeFiles/hydra.dir/build.make:412: CMakeFiles/hydra.dir/src/frontend/frontier_extractor.cpp.o] Error 1
      make[2]: *** Waiting for unfinished jobs....
      make[1]: *** [CMakeFiles/Makefile2:733: CMakeFiles/hydra.dir/all] Error 2
      make: *** [Makefile:136: all] Error 2
      /tmp/pip-build-env-y55rxiga/overlay/lib/python3.8/site-packages/setuptools/command/build_py.py:215: _Warning: Package 'hydra_python.config.uhumans2' is absent from the `packages` configuration.
      !!
      
              ********************************************************************************
              ############################
              # Package would be ignored #
              ############################
              Python recognizes 'hydra_python.config.uhumans2' as an importable package[^1],
              but it is absent from setuptools' `packages` configuration.
      
              This leads to an ambiguous overall configuration. If you want to distribute this
              package, please make sure that 'hydra_python.config.uhumans2' is explicitly added
              to the `packages` configuration field.
      
              Alternatively, you can also rely on setuptools' discovery methods
              (for example by using `find_namespace_packages(...)`/`find_namespace:`
              instead of `find_packages(...)`/`find:`).
      
              You can read more about "package discovery" on setuptools documentation page:
      
              - https://setuptools.pypa.io/en/latest/userguide/package_discovery.html
      
              If you don't want 'hydra_python.config.uhumans2' to be distributed and are
              already explicitly excluding 'hydra_python.config.uhumans2' via
              `find_namespace_packages(...)/find_namespace` or `find_packages(...)/find`,
              you can try to use `exclude_package_data`, or `include-package-data=False` in
              combination with a more fine grained `package-data` configuration.
      
              You can read more about "package data files" on setuptools documentation page:
      
              - https://setuptools.pypa.io/en/latest/userguide/datafiles.html
      
      
              [^1]: For Python, any directory (with suitable naming) can be imported,
                    even if it does not contain any `.py` files.
[... 41 lines cut ...]
          self.run_command(cmd_name)
        File "/tmp/pip-build-env-y55rxiga/overlay/lib/python3.8/site-packages/setuptools/_distutils/cmd.py", line 316, in run_command
          self.distribution.run_command(command)
        File "/tmp/pip-build-env-y55rxiga/overlay/lib/python3.8/site-packages/setuptools/dist.py", line 945, in run_command
          super().run_command(command)
        File "/tmp/pip-build-env-y55rxiga/overlay/lib/python3.8/site-packages/setuptools/_distutils/dist.py", line 989, in run_command
          cmd_obj.run()
        File "/tmp/pip-build-env-y55rxiga/overlay/lib/python3.8/site-packages/setuptools/command/build_ext.py", line 93, in run
          _build_ext.run(self)
        File "/tmp/pip-build-env-y55rxiga/overlay/lib/python3.8/site-packages/setuptools/_distutils/command/build_ext.py", line 359, in run
          self.build_extensions()
        File "/tmp/pip-build-env-y55rxiga/overlay/lib/python3.8/site-packages/setuptools/_distutils/command/build_ext.py", line 479, in build_extensions
          self._build_extensions_serial()
        File "/tmp/pip-build-env-y55rxiga/overlay/lib/python3.8/site-packages/setuptools/_distutils/command/build_ext.py", line 505, in _build_extensions_serial
          self.build_extension(ext)
        File "<string>", line 77, in build_extension
        File "/usr/lib/python3.8/subprocess.py", line 364, in check_call
          raise CalledProcessError(retcode, cmd)
      subprocess.CalledProcessError: Command '['cmake', '--build', '.', '-j4']' returned non-zero exit status 2.
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for hydra_python
Failed to build hydra_python
ERROR: ERROR: Failed to build installable wheels for some pyproject.toml based projects (hydra_python) ```

niyatirawal, 2024-08-02:
After following the instructions in the python bindings section and creating and sourcing a virtual environment, this error was fixed. I am currently facing an error when I try to execute `hydra habitat run /path/to/habitat/mp3d/17DRP5sb8fy/17DRP5sb8fy.glb`. The error is as follows:
`Traceback (most recent call last):
  File "/workspaces/catkin_ws/pyenv/bin/hydra", line 8, in <module>
    sys.exit(main.cli())
  File "/workspaces/catkin_ws/pyenv/lib/python3.8/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
  File "/workspaces/catkin_ws/pyenv/lib/python3.8/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
  File "/workspaces/catkin_ws/pyenv/lib/python3.8/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/workspaces/catkin_ws/pyenv/lib/python3.8/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/workspaces/catkin_ws/pyenv/lib/python3.8/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/workspaces/catkin_ws/pyenv/lib/python3.8/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
TypeError: run() missing 1 required positional argument: 'publish'`

nathanhhughes, 2024-08-05:
Hi, thanks for your interest in our work! As you have seemed to have figured out, the build issue you encountered was gcc running out of memory. For future reference (if you haven't seen it yet), you can set the number of jobs that Hydra uses to compile the bindings by setting the `CMAKE_BUILD_PARELLEL_LEVEL` (see [here](https://github.mit.edu/SPARK/Hydra/blob/63f2585283fcaee4180c6d907fc10fb07a8d180a/setup.py#L65)). 

As per the error w.r.t. running the habitat command: sorry about that, it appears I didn't try that particular command before releasing. I pushed a fix to `main` that will at least fix that, but haven't had a chance to install and test the bindings fully.

niyatirawal, 2024-08-06:
Many thanks for pushing the fix to main.  I wanted to extract scene graphs for the HM3D dataset. 
Upon executing: 
`hydra habitat run /workspaces/abc_ws/kfPV7w3FaU5.glb`, I now get the following error:
```
Renderer: NVIDIA GeForce GTX TITAN X/PCIe/SSE2 by NVIDIA Corporation
OpenGL version: 4.6.0 NVIDIA 535.183.01
Using optional features:
    GL_ARB_vertex_array_object
    GL_ARB_ES2_compatibility
    GL_ARB_separate_shader_objects
    GL_ARB_robustness
    GL_ARB_texture_storage
    GL_ARB_invalidate_subdata
    GL_ARB_texture_storage_multisample
    GL_ARB_multi_bind
    GL_ARB_direct_state_access
    GL_ARB_get_texture_sub_image
    GL_ARB_texture_filter_anisotropic
    GL_KHR_debug
    GL_KHR_parallel_shader_compile
Using driver workarounds:
    no-forward-compatible-core-context
    nv-egl-incorrect-gl11-function-pointers
    no-layout-qualifiers-on-old-glsl
    nv-zero-context-profile-mask
    nv-implementation-color-read-format-dsa-broken
    nv-cubemap-inconsistent-compressed-image-size
    nv-cubemap-broken-full-compressed-image-query
    nv-compressed-block-size-in-bits
[15:05:30:151963]:[Sim] Simulator.cpp(240)::createSceneInstance : Navmesh file location in scene instance : 
[15:05:30:151992]:[Sim] Simulator.cpp(249)::createSceneInstance : Navmesh file not found, checked at filename : ''
[15:05:30:154204]:[Sim] Simulator.cpp(305)::createSceneInstance : Using scene instance-specified Light key : - no_lights -
[15:05:30:154249]:[Sim] Simulator.cpp(415)::instanceStageForSceneAttributes : Start to load stage named : /workspaces/abc_ws/kfPV7w3FaU5.glb with render asset : /workspaces/abc_ws/kfPV7w3FaU5.glb and collision asset : /workspaces/abc_ws/kfPV7w3FaU5.glb
[15:05:32:136250]:[Sim] Simulator.cpp(442)::instanceStageForSceneAttributes : Successfully loaded stage named : /workspaces/abc_ws/kfPV7w3FaU5.glb
[15:05:32:136302]:[Sim] Simulator.cpp(474)::instanceStageForSceneAttributes : 
---
The active scene does not contain semantic annotations : activeSemanticSceneID_ = 0  
---
[15:05:32:136362]:[Sim] Simulator.cpp(208)::reconfigure : CreateSceneInstance success == true for active scene name : /workspaces/abc_ws/kfPV7w3FaU5.glb  with renderer.
[15:05:32:235946]:[Sim] Simulator.cpp(898)::recomputeNavMesh : reconstruct navmesh successful
Traceback (most recent call last):
  File "/workspaces/abc_ws/pyenv/bin/hydra", line 8, in <module>
    sys.exit(main.cli())
  File "/workspaces/abc_ws/pyenv/lib/python3.8/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
  File "/workspaces/abc_ws/pyenv/lib/python3.8/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
  File "/workspaces/abc_ws/pyenv/lib/python3.8/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/workspaces/abc_ws/pyenv/lib/python3.8/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/workspaces/abc_ws/pyenv/lib/python3.8/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/workspaces/abc_ws/pyenv/lib/python3.8/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
  File "/workspaces/abc_ws/pyenv/lib/python3.8/site-packages/hydra_python/commands/habitat.py", line 71, in run
    data = habitat.HabitatInterface(scene_path)
  File "/workspaces/abc_ws/pyenv/lib/python3.8/site-packages/hydra_python/_plugins/habitat.py", line 201, in __init__
    self._make_instance_labelmap()
  File "/workspaces/abc_ws/pyenv/lib/python3.8/site-packages/hydra_python/_plugins/habitat.py", line 231, in _make_instance_labelmap
    names = [name_mapping[ade_to_mpcat[idx]] for idx in keys]
  File "/workspaces/abc_ws/pyenv/lib/python3.8/site-packages/hydra_python/_plugins/habitat.py", line 231, in <listcomp>
    names = [name_mapping[ade_to_mpcat[idx]] for idx in keys]
KeyError: 0
[15:05:32:380711]:[Sim] Simulator.cpp(69)::~Simulator : Deconstructing Simulator
```

niyatirawal, 2024-08-07:
For mp3d scene_dataset, I get `Segmentation fault (core dumped)`. There seems to be something wrong with the colors.
```
Renderer: NVIDIA GeForce GTX TITAN X/PCIe/SSE2 by NVIDIA Corporation
OpenGL version: 4.6.0 NVIDIA 535.183.01
Using optional features:
    GL_ARB_vertex_array_object
    GL_ARB_ES2_compatibility
    GL_ARB_separate_shader_objects
    GL_ARB_robustness
    GL_ARB_texture_storage
    GL_ARB_invalidate_subdata
    GL_ARB_texture_storage_multisample
    GL_ARB_multi_bind
    GL_ARB_direct_state_access
    GL_ARB_get_texture_sub_image
    GL_ARB_texture_filter_anisotropic
    GL_KHR_debug
    GL_KHR_parallel_shader_compile
Using driver workarounds:
    no-forward-compatible-core-context
    nv-egl-incorrect-gl11-function-pointers
    no-layout-qualifiers-on-old-glsl
    nv-zero-context-profile-mask
    nv-implementation-color-read-format-dsa-broken
    nv-cubemap-inconsistent-compressed-image-size
    nv-cubemap-broken-full-compressed-image-query
    nv-compressed-block-size-in-bits
[09:55:43:220206]:[Sim] Simulator.cpp(240)::createSceneInstance : Navmesh file location in scene instance : /workspaces/abc_ws/data/versioned_data/mp3d_example_scene_1.1/17DRP5sb8fy/17DRP5sb8fy.navmesh
[09:55:43:220230]:[Sim] Simulator.cpp(245)::createSceneInstance : Loading navmesh from /workspaces/abc_ws/data/versioned_data/mp3d_example_scene_1.1/17DRP5sb8fy/17DRP5sb8fy.navmesh
[09:55:43:220313]:[Sim] Simulator.cpp(247)::createSceneInstance : Navmesh Loaded.
[09:55:43:245567]:[Sim] Simulator.cpp(305)::createSceneInstance : Using scene instance-specified Light key : - no_lights -
[09:55:43:245610]:[Sim] Simulator.cpp(415)::instanceStageForSceneAttributes : Start to load stage named : /workspaces/abc_ws/data/versioned_data/mp3d_example_scene_1.1/17DRP5sb8fy/17DRP5sb8fy.glb with render asset : /workspaces/abc_ws/data/versioned_data/mp3d_example_scene_1.1/17DRP5sb8fy/17DRP5sb8fy.glb and collision asset : /workspaces/abc_ws/data/versioned_data/mp3d_example_scene_1.1/17DRP5sb8fy/17DRP5sb8fy.glb
[09:55:47:269522]:[Sim] Simulator.cpp(442)::instanceStageForSceneAttributes : Successfully loaded stage named : /workspaces/abc_ws/data/versioned_data/mp3d_example_scene_1.1/17DRP5sb8fy/17DRP5sb8fy.glb
[09:55:47:269586]:[Sim] Simulator.cpp(208)::reconfigure : CreateSceneInstance success == true for active scene name : /workspaces/abc_ws/data/versioned_data/mp3d_example_scene_1.1/17DRP5sb8fy/17DRP5sb8fy.glb  with renderer.
[09:55:47:343102]:[Sim] Simulator.cpp(898)::recomputeNavMesh : reconstruct navmesh successful
[09:55:48:180817]:[Sim] Simulator.cpp(898)::recomputeNavMesh : reconstruct navmesh successful
Trajectory is 105.64831450406255 meters long
I0807 09:55:48.819205 35134 gvd_place_extractor.cpp:88] Downsampling TSDF when creating places!
I0807 09:55:48.845501 35134 hydra_pipeline.cpp:100] 
====================================================================================================
= Modules                                                                                          =
====================================================================================================

****************************************************************************************************
* backend                                                                                          *
****************************************************************************************************
========================================== BackendConfig ===========================================
visualize_place_factors:                     false
enable_rooms:                                true (default)
enable_buildings:                            true (default)
building_color:                              [169, "\b", �, �] (default)
building_semantic_label:                     22 (default)
use_2d_places:                               false (default)
add_places_to_deformation_graph:             true (default)
optimize_on_lc:                              true (default)
enable_node_merging:                         true (default)
merge_update_map:                            {BUILDINGS: false, OBJECTS: false, PLACES: true, ROOMS:
                                             false} (default)
merge_update_dynamic:                        true (default)
places_merge_pos_threshold_m:                0.3
places_merge_distance_tolerance_m:           0.2
use_mesh_subscribers:                        false (default)
enable_merge_undos:                          false (default)
use_active_flag_for_updates:                 true (default)
num_neighbors_to_find_for_merge:             1 (default)
zmq_send_url:                                tcp://127.0.0.1:8001 (default)
zmq_recv_url:                                tcp://127.0.0.1:8002 (default)
use_zmq_interface:                           false (default)
zmq_num_threads:                             2 (default)
zmq_poll_time_ms:                            10 (default)
[... 236 lines cut ...]
   semantic_colormap_file:                   "" (default)
   semantic_label_remap_filepath:            "" (default)
   dynamic_labels:                           [] (default)
   invalid_labels:                           [0]
   object_labels:                            [3, 5, 6, 7, 8, 10, 11, 12, 13, 14, 15, 18, 19, 20, 21,
                                             22, 23, 25, 27, 28, 30, 31, 32, 33, 34, 35, 36, 37]
   surface_places_labels:                    [2, 16]
[... cut ...]
    @     0x7fce6d39b623 flann::anyimpl::small_any_policy<>::print()
    @     0x7fce65fee6ad flann::get_param<>()
    @     0x7fce66037642 pcl::KdTreeFLANN<>::setInputCloud()
    @     0x7fce660323aa pcl::search::KdTree<>::setInputCloud()
    @     0x7fce6602e818 hydra::findClusters()
    @     0x7fce6602ef9d hydra::MeshSegmenter::detect()
    @     0x7fce65face6a hydra::FrontendModule::updateObjects()
    @     0x7fce89846df4 (unknown)
    @     0x7fce9ee59609 start_thread
    @     0x7fce9ef93353 clone
Segmentation fault (core dumped)
```