MIT-SPARK/Hydra #63: [QUESTION] undefined reference to `YAML::detail::node_data::empty_scalar[abi:cxx11]()'

lyxichigoichie, 2024-08-16:
while I was building `kimera_pgmo_ros` package, it always report 
```
/usr/bin/ld: /home/lyxichigoichie/Documents/Scene_Graph/Hydra/devel/lib/libconfig_utilities.so: undefined reference to `YAML::detail::node_data::empty_scalar[abi:cxx11]()
```
 and then failed to build the project. Anyone has the same problem?

OS: Ubuntu20
ROS: noetic

zszz3, 2024-12-02:
Hello，i meet the same problem，do you solve your problem？

Walklism, 2024-12-14:
I encountered the same kind of problem and found that the culprit had something to do with **using the Conda Virtual Environment**. 
I somehow solved it by commenting out the lines "export PATH=~/miniconda3/bin:$PATH" and "source ~/miniconda3/bin/activate" in the ~/.bashrc file (which I had added for convenience), and then restarting the terminal.

*_(My error log in the terminal was extremely long, comprising probably hundreds of lines of error reports, but almost all of them followed the format /usr/bin/ld: ..... : undefined reference to YAML::detail:: ......)_

OS: Ubuntu 20.04 in VMWare
ROS distro: Noetic
Conda used: Miniconda

Walklism, 2024-12-14:
I encountered the same kind of problem and my guess is that the culprit is **using Conda Virtual Environment**, because
I somehow solved it by commenting out the lines like "export PATH=~/miniconda3/bin:$PATH" and "source ~/miniconda3/bin/activate" in the ~/.bashrc file (which I had added for convenience), and then restarting the terminal.

*(My error log in the terminal was extremely long, comprising probably hundreds of lines of error reports, but almost all of them followed the format _'/usr/bin/ld: ..... : undefined reference to YAML::detail:: ......'_ , which applies to the issue.)

OS: Ubuntu 20.04 in VMWare
ROS distro: Noetic
Conda used: Miniconda

Walklism, 2024-12-14:
I encountered the same kind of problem when building the **config_utilities** package, and the culprit for me was **adding the Conda virtual environment to PATH**. 
I solved the problem by simply commenting out the line export `PATH=~/miniconda3/bin:$PATH` in the ~/.bashrc file (which I had added for convenience) and then restarting the terminal.
Additionally, the problem can be reproduced by executing export `PATH=~/miniconda3/bin:$PATH` in the terminal and build again.

Here is a small piece of my error log in the terminal. The full log is extremely long, comprising probably hundreds of lines of error reports, but almost all of them follow the format:
`/usr/bin/ld: ..... : undefined reference to 'YAML::detail::......'`
which applies to the current issue.
[image]

OS: Ubuntu 20.04 in VMware
ROS distro: Noetic
Conda used: Miniconda

zszz3, 2024-12-14:
> I encountered a similar problem when building the **config_utilities** package, and the culprit for me was **adding the Conda virtual environment to PATH**. I solved the problem by simply commenting out the line `export PATH=~/miniconda3/bin:$PATH` in the ~/.bashrc file (which I had added for convenience) and then restarting the terminal. Additionally, the problem can be reproduced by executing `export PATH=~/miniconda3/bin:$PATH` in the terminal and rebuilding the package.
> 
> The image below is a piece of my error log in the terminal. The full log is extremely long, comprising probably hundreds of lines of error reports, but almost all of them follow the format: `/usr/bin/ld: ..... : undefined reference to 'YAML::detail::......'` which also applies to the current issue. I hope these information proves helpful to someone. ![myerror](https://private-user-images.githubusercontent.com/128035971/395779205-c737077e-aa96-46a4-a743-646ad4f90592.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQxODQ3NjMsIm5iZiI6MTczNDE4NDQ2MywicGF0aCI6Ii8xMjgwMzU5NzEvMzk1Nzc5MjA1LWM3MzcwNzdlLWFhOTYtNDZhNC1hNzQzLTY0NmFkNGY5MDU5Mi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjE0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIxNFQxMzU0MjNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wZDI0MDk2MzY5ZDczNzg1YTFjMjY5ZTY3Njc3MWE0MjgyNzRlZmMxMzU5MzY1N2ZmZTQwZmY1ZjE2ODg0ZDIwJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.d6lSng8YWlLefPWkI4np0n9v3Gf_vvR636mm_uiWflc)
> 
> OS: Ubuntu 20.04 in VMware ROS distro: Noetic Conda used: Miniconda

Thank you for your help. With your help, I solved this problem! I wish you good luck in your research.