MIT-SPARK/Hydra #69: [BUG] /ros/assert.h:130:9: error: no match for ‘operator!’

PengZai, 2024-11-07:
**Describe the bug**
A clear and concise description of what the bug is.

hi guys, I came across a problem when I run catkin build. There is an error on the process of compiling kimera_pgmo_rviz. it said 

Errors     << kimera_pgmo_rviz:make /home/zhipeng/vscode_projects/catkin_ws/logs/kimera_pgmo_rviz/build.make.000.log
In file included from /home/zhipeng/vscode_projects/catkin_ws/src/kimera_pgmo/kimera_pgmo_rviz/src/mesh_visual.cpp:12:
/home/zhipeng/vscode_projects/catkin_ws/src/kimera_pgmo/kimera_pgmo_rviz/src/mesh_visual.cpp: In member function ‘Ogre::Material& kimera_pgmo::MeshVisual::getMaterial() const’:
**/opt/ros/noetic/include/ros/assert.h:130:9: error: no match for ‘operator!’ (operand type is ‘Ogre::SharedPtr<Ogre::Material>’)**

how can I fix it? please help me.

by the way, I compile this project based on Ubuntu 20.04 of WSL(windows subsystem for Linux)

[image]

**System Information:**
 - OS [e.g. Ubuntu 18.04, Ubuntu 20.04]: 
 - Any build warnings? [yes/no]: 
 - Any non-system dependency versions that you know of? [yes/no]: 

**Additional Information:**
If possible and/or relevant, please take a look [here](/MIT-SPARK/Hydra/blob/main/doc/debugging.md) and try to obtain the following (attached as files preferrably):
- A backtrace (either from gdb or from Hydra directly)
- The current workspace configuration
- The current workspace contents

zszz3, 2024-12-02:
Hello,I meet the same problem,do you solve this problem?

PengZai, 2024-12-09:
> Hello,I meet the same problem,do you solve this problem?

if you dont need debug， then change the setting to release， it could be pass then

closed 2026-09-09 by nathanhhughes