MIT-SPARK/Hydra #72: [QUESTION]Use another VIO/SLAM pose source

rookierobot, 2024-11-28:
Hi, I've read the discussion on 1 and successfully used my own semantic segmentation network to replace the "seg image" in the uHumans dataset to launch hydra, and since my robot doesn't have an odometer, I'm going to use Orb-SLAM3 as the external visual odometer. And I want to use the realsensed455 camera to run hydra indoor in real time. In the launch file, I modified 'use_gt_frame'/ 'robot_frame '/' odom_frame '/ 'sensor_frame' in the part about the mileage meter, but it doesn't seem to work. The following is my launch file, do I miss any parameters?
`<launch>
    <arg name="sim_time_required" default="false"/>
    <param name="use_sim_time" value="$(arg sim_time_required)"/>

    <arg name="config_dir" default="$(find hydra)/config/d455_outdoor"/>
    <arg name="use_gt_frame" default="false" doc="use simulator-provided tf frame"/>
    <arg name="use_gt_semantics" default="false" doc="use simulator-provider semantics"/>
    <arg name="use_static_tfs" default="true" doc="publish static tfs from file"/>
    <arg name="color_mesh_by_label" default="false" doc="display mesh colored by semantic label"/>
    <arg name="use_oriented_bounding_boxes" default="false"/>
    <arg name="use_2d_places" default="true"/>
    <arg name="use_single_channel_label_image" default="false"/>

    <arg name="robot_frame" default="$(eval 'base_link_gt' if arg('use_gt_frame') else 'camera_link')" doc="robot body tf frame"/>
    <arg name="odom_frame" default="$(eval 'world' if arg('use_gt_frame') else 'map')" doc="odometry (map) frame"/>
    <arg name="sensor_frame" default="$(eval 'left_cam' if arg('use_gt_frame') else 'camera_color_frame')" doc="camera frame"/>

    
    <arg name="labelspace_name" default="$(eval 'uhumans2_office' if arg('use_gt_semantics') else 'ade20k_indoor')" doc="semantic label space"/>
    <arg name="semantic_map_dir" default="$(find hydra_ros)/config/color" if="$(arg use_gt_semantics)"/>
    
    
    <arg name="semantic_map_path" default="/home/robot/hydra_ws/src/hydra_ros/hydra_ros/config/color/ade20k_indoor.csv"/>
    <arg name="semantic_label_remap_filepath" default="$(find hydra)/config/label_remaps/uhumans2_office.yaml" if="$(arg use_single_channel_label_image)"/>

    <arg name="rgb_topic" default="/camera/color/image_raw"/>
    <arg name="rgb_info_topic" default="/camera/aligned_depth_to_color/camera_info"/>
    <arg name="depth_topic" default="/camera/aligned_depth_to_color/image_raw"/>
    <arg name="label_topic" default="/mmsegmentation/rgb" unless="$(arg use_gt_semantics)"/>
    <arg name="label_topic" default="$(eval '/tesse/seg_cam/converted/image_raw' if arg('use_single_channel_label_image') else '/tesse/seg_cam/rgb/image_raw')" 
                            if="$(arg use_gt_semantics)"/>

    <arg name="prerecorded_topic" default="/oneformer/labels/image_raw"/>

    <include file="$(find hydra_ros)/launch/static_tfs/uhumans2_static_tfs.xml" if="$(arg use_static_tfs)"/>

    <node pkg="tf2_ros" type="static_transform_publisher" name="fake_world_tf" args="0 0 0 0 0 0 1 map world" if="$(arg use_gt_frame)"/>

    

    <include file="$(find hydra_ros)/launch/hydra.launch" pass_all_args="true">
        <arg name="dataset_name" default="uhumans2"/>
        <arg name="rviz_file" default="river.rviz"/>
    </include>

</launch>
`
[image]
thanks for your reply!

rookierobot, 2024-11-28:
[image]
From the rqt_graph, everything seems to be fine

rookierobot, 2024-11-29:
as i said, i change frame parameters in the launch file and launch the hydra but i got the following infos:
`W1129 10:03:08.431362 68297 pgmo_glog_sink.cpp:51] [kimera_pgmo_interface.cpp:516] KimeraPgmo: Partial mesh not connected to pose graph.
W1129 10:03:08.834178 68297 pgmo_glog_sink.cpp:51] [kimera_pgmo_interface.cpp:516] KimeraPgmo: Partial mesh not connected to pose graph.
[ WARN] [1732845790.310794373, 36.651787235]: Detected jump back in time of 0.00309276s. Clearing TF buffer.
[ WARN] [1732845790.310802312, 36.651787235]: Detected jump back in time of 0.00309276s. Clearing TF buffer.
E1129 10:03:10.759598 68296 lookup_tf.cpp:95] Failed to find: map_T_camera_link_slam @ 36604880000 [ns]: Lookup would require extrapolation at time 36.604880000, but only time 36.654879999 is in the buffer, when looking up transform from frame [camera_link_slam] to frame [map]. canTransform returned after 0 timeout was 0.
W1129 10:03:10.759712 68296 input_module.cpp:114] [Hydra Input] dropping input @ 36604880000 [ns] due to missing pose
W1129 10:03:24.655104 68297 pgmo_glog_sink.cpp:51] [kimera_pgmo_interface.cpp:516] KimeraPgmo: Partial mesh not connected to pose graph.
W1129 10:03:24.752213 68297 pgmo_glog_sink.cpp:51] [kimera_pgmo_interface.cpp:516] KimeraPgmo: Partial mesh not connected to pose graph.
W1129 10:03:24.912735 68297 pgmo_glog_sink.cpp:51] [kimera_pgmo_interface.cpp:516] KimeraPgmo: Partial mesh not connected to pose graph.
W1129 10:03:26.280938 68297 pgmo_glog_sink.cpp:51] [kimera_pgmo_interface.cpp:516] KimeraPgmo: Partial mesh not connected to pose graph.
[ WARN] [1732845806.474233113, 39.100064603]: Detected jump back in time of 0.0048154s. Clearing TF buffer.
[ WARN] [1732845806.474246226, 39.100064603]: Detected jump back in time of 0.0048154s. Clearing TF buffer.
E1129 10:03:26.919359 68296 lookup_tf.cpp:95] Failed to find: map_T_camera_link_slam @ 39054879999 [ns]: Lookup would require extrapolation at time 39.054879999, but only time 39.104880000 is in the buffer, when looking up transform from frame [camera_link_slam] to frame [map]. canTransform returned after 0 timeout was 0.
W1129 10:03:26.919409 68296 input_module.cpp:114] [Hydra Input] dropping input @ 39054879999 [ns] due to missing pose`

rookierobot, 2024-11-29:
i got the info in terminial:
` [kimera_pgmo_interface.cpp:516] KimeraPgmo: Partial mesh not connected to pose graph. `
and the rviz :
[image]

closed 2024-12-07 by rookierobot

maffan-96, 2025-05-06:
> i got the info in terminial: `[kimera_pgmo_interface.cpp:516] KimeraPgmo: Partial mesh not connected to pose graph.` and the rviz : [image]

Were you able to resolve that issue? How? (I am facing same issue)