MIT-SPARK/Hydra #76: [QUESTION] Encountered `zmq::error_t` when executing `roslaunch`.

Speike-SHL, 2025-02-14:
Hello, I've successfully completed the build using `catkin build`. However, when I attempted to start the system with the command `roslaunch hydra_ros uhumans2.launch`, I encountered the following error. I'm hoping you can provide a solution. 

```bash
setting /run_id to e2f093be-eb11-11ef-a5c4-6faa4d339ca2
process[rosout-1]: started with pid [489753]
started core service [/rosout]
process[bag_static_tf_0-2]: started with pid [489760]
process[bag_static_tf_1-3]: started with pid [489761]
process[bag_static_tf_2-4]: started with pid [489762]
process[bag_static_tf_3-5]: started with pid [489763]
process[bag_static_tf_4-6]: started with pid [489772]
process[bag_static_tf_5-7]: started with pid [489779]
process[fake_world_tf-8]: started with pid [489785]
process[hydra_ros_node-9]: started with pid [489792]
process[hydra_dsg_visualizer-10]: started with pid [489794]
process[rviz-11]: started with pid [489803]
I0215 04:25:53.456243 489792 hydra_ros_pipeline.cpp:77] Starting Hydra-ROS with input configuration
====================================== RosInputModule::Config ======================================
clear_queue_on_fail:                         true (default)
inputs[left_cam] [InputModule::InputPair::Config] (default):
   receiver [Virtual Config: ClosedSetImageReceiver::Config]:
      input_separation_s:                    0 (default)
      ns:                                    "" (default)
      queue_size:                            10 (default)
   sensor [Virtual Config: RosCamera::Config]:
      min_range [m]:                         0.1
      max_range [m]:                         5
      ns:                                    "" (default)
      extrinsics [Virtual Config: RosExtrinsics::Config]:
         sensor_frame:                       left_cam
tf_lookup [TFLookup::Config] (default):
   wait_duration_s:                          0.1 (default)
   buffer_size_s:                            30 (default)
   max_tries:                                5 (default)
   verbosity:                                3 (default)
====================================================================================================
I0215 04:25:53.545698 489792 gvd_place_extractor.cpp:92] Downsampling TSDF when creating places!
terminate called after throwing an instance of 'zmq::error_t'
  what():  Address already in use
*** Aborted at 1739564753 (unix time) try "date -d @1739564753" if you are using GNU date ***
PC: @     0x7ff2e082e00b gsignal
*** SIGABRT (@0x3e800077940) received by PID 489792 (TID 0x7ff2cf5d7a40) from PID 489792; stack trace: ***
    @     0x7ff2e0c8c631 (unknown)
    @     0x7ff2e082e090 (unknown)
    @     0x7ff2e082e00b gsignal
    @     0x7ff2e080d859 abort
    @     0x7ff2e0aa6ee6 (unknown)
    @     0x7ff2e0ab8f8c (unknown)
    @     0x7ff2e0ab8ff7 std::terminate()
    @     0x7ff2e0ab9258 __cxa_throw
    @     0x7ff2e171db67 _ZN9spark_dsg9ZmqSenderC2ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEm.cold
    @     0x7ff2e1a3cdad hydra::ZmqSink::ZmqSink()
    @     0x7ff2e220a5e0 hydra::HydraRosPipeline::init()
    @     0x55f2b5b11eed main
    @     0x7ff2e080f083 __libc_start_main
    @     0x55f2b5b11ffe _start
================================================================================REQUIRED process [hydra_ros_node-9] has died!
process has died [pid 489792, exit code -6, cmd /home/li/shl/Hydra_ws/devel/lib/hydra_ros/hydra_ros_node --minloglevel=0 -v=0 hydra_ros_node/input/left_cam/depth_registered/image_rect:=/tesse/depth_cam/mono/image_raw hydra_ros_node/input/left_cam/rgb/image_raw:=/tesse/left_cam/rgb/image_raw hydra_ros_node/input/left_cam/rgb/camera_info:=/tesse/left_cam/camera_info hydra_ros_node/input/left_cam/semantic/image_raw:=/tesse/seg_cam/rgb/image_raw __name:=hydra_ros_node __log:=/home/li/.ros/log/e2f093be-eb11-11ef-a5c4-6faa4d339ca2/hydra_ros_node-9.log].
log file: /home/li/.ros/log/e2f093be-eb11-11ef-a5c4-6faa4d339ca2/hydra_ros_node-9*.log
Initiating shutdown!
================================================================================
```

Walklism, 2025-03-11:
I've encountered a similiar problem while executing roslaunch hydra_ros uhumans2.launch, where **"what():  Address already in use"** also appears. My problem source was VScode automatically forwarding port, therefore causing conflict. 
I solved it by disabling the option `remote.autoForwardPorts` in VScode settings. 
Hope this helps.

[image]

cho-jang-hyun, 2025-03-18:
In my case, I resolved the issue as follows:  

Hydra uses `ZeroMQ` to bind to `tcp://127.0.0.1:8001` and `tcp://127.0.0.1:8002`. However, I found that these ports were already in use, either by the host PC or due to some other process in my environment, causing a conflict.  

To resolve this, I checked in host for port usage with the following command:  

```sh
sudo netstat -ltnp
```  

If the ports were occupied, I terminated the conflicting process and then re-ran hydra  launch, which worked successfully.  

```sh
kill -9 <pid>
```  

Hope this helps others facing the same issue! 🚀

Speike-SHL, 2025-03-19:
The problem has been resolved. Thanks a lot!

closed 2025-03-19 by Speike-SHL