MIT-SPARK/Hydra #77: [BUG] RuntimeError: Invalid config 'Sensor': Fatal: Check [3/3] failed: Virtual config 'extrinsics' is not set and not marked optional.

LeonGoretzkatju, 2025-02-15:
**Describe the bug**
A clear and concise description of what the bug is.
Thanks for your excellent work! I was trying to use the python binding version of Hydra, and when I execute the following command based on your readme file:
`hydra run ~/data/scene_datasets/mp3d/1LXtFkjw3qL/1LXtFkjw3qL.glb --openset-model "ViT-L/14" -o ~/test_python_clip`
My system report the following error:

[image]
Looks like there is some problem during the python binding process, Thank you!
**System Information:**
 - OS [Ubuntu 20.04]: 
 - Any build warnings? [no]: 
 - Any non-system dependency versions that you know of? [no]: 

**Additional Information:**
If possible and/or relevant, please take a look [here](/MIT-SPARK/Hydra/blob/main/doc/debugging.md) and try to obtain the following (attached as files preferrably):
- A backtrace (either from gdb or from Hydra directly)
- The current workspace configuration
- The current workspace contents

psi0000, 2025-02-20:
I have the same problem too. 
I wonder it's solution

hys02245, 2025-03-25:
same problem, too